lancaster-university/codal-microbit-v2 #4: Some refactoring of SoundExpressions

pelikhan, 2020-10-25:
Suggestion to the current implementation:

- [ ] use a binary format that can be mapped to a C struct
- [ ] removing all parsing code, use of ManagedString
- [ ] convert existing sounds to binary format

Once this is done, we can lift the sound definitions to MakeCode resources so that the simulator can eventually play them.

finneyj, 2020-10-25:
The underlying SoundEffect used by the Synth is a binary format that I think would suit if we get a nice JS editor to create sound effects.

I have a ticket to write a frequency slide effect that would reduce most of the current "SoundEmojis" to one or two binary SoundEffect frames. Once we have that, would the SoundEffect structure (and friends) suit @pelikhan ?

https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/SoundEmojiSynthesizer.h#L94

pelikhan, 2020-10-25:
Yes it looks better. Eventually we should be able to build an editor for it. What is the relation of soundeffect to the previous CODAL sound pipeline?

https://github.com/lancaster-university/codal-microbit-v2/blob/master/inc/SoundEmojiSynthesizer.h#L94<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Flancaster-university%2Fcodal-microbit-v2%2Fblob%2Fmaster%2Finc%2FSoundEmojiSynthesizer.h%23L94&data=04%7C01%7Cjhalleux%40microsoft.com%7C5cea22c0e137425528f008d8790e4cbf%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637392449608085470%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=ETY6bQnoG1KoUFd5Al9IO9GZFdFuIKDWN01Z%2BpJDH8k%3D&reserved=0>

finneyj, 2020-10-26:
Great - thanks @pelikhan. I thought you might like to write an editor for this in due course. That would give a great experience I think. I'd like to upstream this audio pipeline once we've stabilized it here I think (and replace the Mixer/Synthesizer in codal-core)... so this could also be useful for a future version of arcade etc.

The SoundExpressions class is just a thin wrapper that creates a series of SoundEffect structures from a String serialization. @microbit-matt-hillsdon  wrote that for us to interface with a JS design tool that some sound engineers have been working with for @jaustin.

Audio output pipeline looks like this:

[image]

jaustin, 2020-10-26:
Definitely up for a nice editor for these synth sounds (we did a tool for the sound designer to use and it was really fun) . I think a binary serialisation makes a lot of sense, but removing all parsing code from the DAL is not a good answer because it means we don't have consistency across editors for sound parsing - one of the key reasons CODAL exists. Would be happy for DAL to assume correctly-formed binary serialisation as long as that is defined in CODAL. 

Whatever the serialisation format, it needs to be possible for a cool sound made to sound the same in any editor building on CODAL. This is part of the promise for people using CODAL is that they get al micro:bit features, and this is a key feature The Foundation has been designing for v2.

We plan to continue to evolve and add to the sounds, and propogating them out from the DAL is the best way to do this, so having a set of built-in sounds remains important.

@dpgeorge at the moment you're just using the built-in sounds, right? Any requirements around serialisation format from your end?

dpgeorge, 2021-02-14:
> at the moment you're just using the built-in sounds, right? Any requirements around serialisation format from your end?

At the moment MicroPython exposes them like `audio.play(Sound.HAPPY)` and only supports built-in sounds.  There are therefore no requirements for serialisation.