lancaster-university/codal-microbit-v2 #6: Ability to stop a sound expression

pelikhan, 2020-10-28:
We need to be able to button-smash a button and play the latest sound expression.

jaustin, 2020-10-28:
This is nice for ‘button bashing’ and avoids queues. It’s also the behaviour I’d expect if we were playing back a sample so you can make a sampler that plays back the ‘sample’ ’s s s s sample sam sam sample’ if you press the button at the right time.

finneyj, 2020-10-28:
But you might also want the queuing behaviour... The ability to back-to-back sounds also has applications.

Would it make sense to follow the design pattern we have for other long running tasks, like scrolling/animating the display:
https://github.com/lancaster-university/codal-core/blob/master/source/drivers/AnimatedDisplay.cpp#L252

Here we have an explicit abort... the cancels any active animation and moves onto the next one. 

I'm not sure I agree this behaviour is what you'd expect for playing back a sample. it' snot like you say 

print ("Jonny");
print ("Jonny");
print ("Jonny");

and expect J-J-Jonny... 

:)

pelikhan, 2020-10-28:
Having a block to abort any sound works too.

https://github.com/lancaster-university/codal-core/blob/master/source/drivers/AnimatedDisplay.cpp#L252<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Flancaster-university%2Fcodal-core%2Fblob%2Fmaster%2Fsource%2Fdrivers%2FAnimatedDisplay.cpp%23L252&data=04%7C01%7Cjhalleux%40microsoft.com%7Ce7ae4f25ddf64f75809d08d87b777cdc%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637395100407891641%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2BFGZfXJuMrPlgaTRw%2FF2Il2VXla4IiOog3K1OuCdwVY%3D&reserved=0>

finneyj, 2020-10-30:
Support for this added as of v0.2.15...
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/SoundExpressions.cpp#L103

closed 2020-11-07 by finneyj