lancaster-university/codal-microbit-v2 #8: I2C read issues in MakeCode/CODAL without a repeat start

microbit-carlos, 2020-10-28:
This simple MakeCode I2C write-read transaction, extracted from the Gigglebot extension, works correctly in v1 (in my device it reads "5"), but returns "0" on v2:

```ts
enum gigglebotI2CCommands {
  GET_FIRMWARE_VERSION = 1,
  GET_MANUFACTURER,
  GET_BOARD,
  GET_VOLTAGE_BATTERY,
  GET_LINE_SENSORS,
  GET_LIGHT_SENSORS,
  GET_MOTOR_STATUS_RIGHT,
  GET_MOTOR_STATUS_LEFT,
  SET_MOTOR_POWER,
  SET_MOTOR_POWERS
}
let ADDR = 0x04

function firmwareVersion(): number {
    let buf = pins.createBuffer(1)
    buf.setNumber(NumberFormat.UInt8BE, 0, gigglebotI2CCommands.GET_FIRMWARE_VERSION)
    pins.i2cWriteBuffer(ADDR, buf)
    let val = pins.i2cReadBuffer(ADDR, 2)
    basic.showNumber(val.getNumber(NumberFormat.Int8BE,0))
    basic.showIcon(IconNames.Heart)
    basic.showNumber(val.getNumber(NumberFormat.Int8LE,0))
    basic.showIcon(IconNames.Heart)
    return val.getNumber(NumberFormat.UInt16BE, 0);
}

basic.pause(1000)
basic.showNumber(firmwareVersion())
```

- [microbit-manual-fw-version-read.hex.zip](https://github.com/microsoft/pxt-microbit/files/5454357/microbit-manual-fw-version-read.hex.zip)

On a v1 we can clearly see the write/read:
[image]

Looking at the clock signal there are two transaction:
1) Write 1 byte (2 byes)
    1) 0x8 (Address 0x4 write)
    2) 0x1 (command to read firmware version)
2) Read 2 bytes (3 bytes in total)
    1) 0x9 (Address 0x4 read)
    2) 0x0 (MSB)
    3) 0x5 (LSB)

With the v2 connected we can see the second part is not complete :

1) Write 1 byte (2 byes total)
    1) 0x8 (Address 0x4 write)
    2) 0x1 (command to read firmware version)
2) Read 2 bytes (3 bytes in total)
    1) 0x9 (Address 0x4 read)
    - Transmission ends

Looks like the I2C device is sending a NACK to the read command (not pulling down the 9th bit)?
Not sure, will need to look at I2C spec.

[image]

This DAL/CODAL programme works on both v1 and v2:
- [dal-repeat-start.hex.zip](https://github.com/microsoft/pxt-microbit/files/5454508/dal-repeat-start.hex.zip)
- [codal-repeat-start.hex.zip](https://github.com/microsoft/pxt-microbit/files/5454513/codal-repeat-start.hex.zip)

```cpp
#include "MicroBit.h"

MicroBit uBit;

enum gigglebotI2CCommands {
  GET_FIRMWARE_VERSION = 1,
  GET_MANUFACTURER,
  GET_BOARD,
  GET_VOLTAGE_BATTERY,
  GET_LINE_SENSORS,
  GET_LIGHT_SENSORS,
  GET_MOTOR_STATUS_RIGHT,
  GET_MOTOR_STATUS_LEFT,
  SET_MOTOR_POWER,
  SET_MOTOR_POWERS
};

static const uint16_t ADDR = 0x04 << 1;

int firmwareVersion() {
    int result;

    uint8_t command = GET_FIRMWARE_VERSION;
    result = uBit.i2c.write(ADDR, &command, 1, true);
    if (result != MICROBIT_OK) {
        uBit.display.print(result);
        return -1;
    }

    uint8_t buffer[2] = {0xFF, 0xFF};
    result = uBit.i2c.read(ADDR, buffer, 2);
    if (result != MICROBIT_OK) {
        uBit.display.print(result);
        return -2;
    }
    return buffer[1] | (buffer[0] << 8);
}

int main() {
    int v = firmwareVersion();

    while(1) {
        uBit.display.print(v);
        uBit.sleep(1000);
        uBit.display.clear();
        uBit.sleep(1000);
    }
}
```

But changing the repeat start from `true` to `false` breaks the programme in v2 with CODAL, but not in v1 with DAL.

I haven't checked the MakeCode source code in depth, but it looks like it is not sending the repeat start unless explicitly requested:
https://github.com/microsoft/pxt-microbit/blob/c3782db37081e36f211ef0637cc05e691071c773/libs/core/pins.cpp#L464

Tested in MakeCode:
makecode.microbit.org version:  3.1.20
Microsoft MakeCode version:  6.6.3
microbit runtime version:  v2.2.0-rc6

With micro:bits:
- v1.3
- v2 RC1

finneyj, 2020-10-29:
Thanks @microbit-carlos - this is a really great analysis.

Curious... The only difference I can perceive in the non-repeated start case is that the v1 DAL left the I2C bus idle for longer before initiating the second half of the transaction...  would you agree @microbit-carlos ?

microbit-carlos, 2020-11-03:
Another possible extension with a similar issue (still need to check if repeated starts work): https://github.com/microsoft/pxt-microbit/issues/3531

microbit-carlos, 2021-02-08:
CODAL I2C pins config (read via debugger), PIN_CNF[0_08] and PIN_CNF[0_16]:
- `0x0000000c`
- DIR `0b0` -> Configure pin as an input pin
- INPUT `0b0` -> Connect input buffer
- PULL `0b11` -> Pullup
- DRIVE `0b0000` -> S0S1 (Standard '0', standard '1')
- SENSE `0b00` -> Disabled

[image]

- samples commit: https://github.com/lancaster-university/microbit-v2-samples/commit/1b416f43df845694222f7e9717989fc538ee0db9
- codal-microbit-v2  commit: 2d8a6d4bf9bec22dba548633df2ae821861b29da
- codal-nrf52: https://github.com/lancaster-university/codal-nrf52/commit/6372c268567a5a75e72db0fdf09e088eb8738e12
- codal-core  commit: https://github.com/lancaster-university/codal-core/commit/969813d451c33daeccef3b41cf904f6c08e6974f
- codal-microbit-nrf5sdk commit:  https://github.com/microbit-foundation/codal-microbit-nrf5sdk/commit/691869a8261e08b283deb0e2267fae3c00c17dae

microbit-carlos, 2021-02-09:
I2C from CODAL without modifications:
- samples commit: https://github.com/lancaster-university/microbit-v2-samples/commit/1b416f43df845694222f7e9717989fc538ee0db9
- codal-microbit-v2  commit: 2d8a6d4bf9bec22dba548633df2ae821861b29da
- codal-nrf52: https://github.com/lancaster-university/codal-nrf52/commit/6372c268567a5a75e72db0fdf09e088eb8738e12
- codal-core  commit: https://github.com/lancaster-university/codal-core/commit/969813d451c33daeccef3b41cf904f6c08e6974f
- codal-microbit-nrf5sdk commit:  https://github.com/microbit-foundation/codal-microbit-nrf5sdk/commit/691869a8261e08b283deb0e2267fae3c00c17dae

## V2 delay

Using the same C++ example with the Gigglebot.

Normal I2C code fails by the device Nacking:
[image]

Zooming in we can see the bus free time (from the first transaction stopping and the next starting) is about 10us
[image]

Adding an 11us delay between the transactions makes it work on a warm restart (pressing the reset button), but not from a cold start. For a cold start it needs more delays, but that looks to be a device initialisation time, so let's ignore that for now.

With a 11us delay the total bus free time becomes about 22us (I can also confirm that with a 10us delay, resulting in total bus free time of 21us, it continues to Nack):
[image]
[image]

## V1 speed up

In V1 normal succesfful run with the same code has a bus free time of 25us:
[image]
[image]

Now on V1 if I modify the I2C code to reduce the time between transaction we also get the same NACK when the bus free time is 20us:
[image]
[image]

Adding a small delay to a bus free time of 22us fixes it here as well:
[image]
[image]

## This results might be device specific

The findings described here might be specific to the bus free time requirements for this accessory (it's an atmega328p with custom software to act as an i2c peripheral), while mcu datasheet doesn't indicate any special requirements, this can be very dependant on it firmware.

Some I2C devices, like the AT42QT2160 in the Klef Piano actually have a bus free time requirement of 50us in the [datasheet](https://www.microchip.com/wwwproducts/en/AT42qt2160), so that explains why it worked on V1 and not on V2 without the repeated start.

I should do the same test with another I2C device, but don't have anything specific at hand, I'll need to look around.

finneyj, 2021-02-10:
Thanks @microbit-carlos - excellent analysis. This seems to confirm our working hypothesis that this compatibility issues are indeed related to the duration of bus free time between transactions.

As you say, this will be highly device dependent. perhaps we'll need to introduce a configurable delay between transactions to allow peripheral manufacturers to account for their requirements. We could also introduce a CONFIG options for that delay, so we can define default value on a per target basis (i.e have something more forgiving than the I2C spec for micro:bit)?

I have also applied a few small improvements to the I2C driver that I have on a branch. I have an Klef Piano inbound (eta in a few hours), so will try a native c++ test against that to see if those changes help.