lancaster-university/codal-microbit-v2 #12: V2 calibration does not persist on power cycles

microbit-mark, 2020-11-03:
**Describe the bug**
The compass calibration should persist through a power cycle/reset. This works on V1 boards, but not V2.

**To Reproduce**
- Flash file the uses the compass
- calibrate
- power cycle

**Expected behavior**
Calibration persists unless flashed using drag and drop

See issue in MakeCode https://github.com/microsoft/pxt-microbit/issues/3550 this reproduces in Python beta too

microbit-carlos, 2020-11-03:
Where is the calibration recorded? In the MicrobitStorage space at 0x74000 or Flash Data Storage at 0x75000? If it goes anywhere else we need to reflect it in the memory map, so that it doesn't clash with things like the MicroPython filesystem.

martinwork, 2020-11-03:
It possibly isn't stored anywhere at present. I believe it goes in MicroBitStorage in v1 DAL, but I think the ultimate plan for CODAL is to use I2C flash.

finneyj, 2020-11-16:
Introduced in v0.2.2.0

microbit-mark, 2020-11-25:
This is now working so I think we can close

closed 2020-11-25 by microbit-mark