lancaster-university/codal-microbit-v2 #14: display readLightLevel always returns 0

microbit-carlos, 2020-11-05:
This example programme always reads 0:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    while (1) {
        uBit.serial.send(ManagedString(uBit.display.readLightLevel()) + "\r\n");
        uBit.sleep(250);
    }
}
```

finneyj, 2020-11-05:
thanks @microbit-carlos. This certainly used to work - must be a regression... I'll take a look.

microbit-mark, 2020-11-06:
Seems to be different behaviour on v1 and v2 for light level. On v2 the initial reading is 0 and on v1 still 255 as documented in 
https://github.com/lancaster-university/microbit-dal/issues/426

finneyj, 2020-11-07:
Fixed in master for v2.

@microbit-mark - I think we fixed that bug on v1? I seem to remember a contribution for that...

microbit-carlos, 2020-11-09:
Just to clarify, the issue wasn't about the first value, as in my example programme the value returned by `uBit.display.readLightLevel()` was **always** `0`.

finneyj, 2020-11-09:
It takes a fair bit of light to trip it...

finneyj, 2020-11-09:
as did a v1....

microbit-carlos, 2020-11-09:
When I was testing this last week I had my phone flash on top of the display and it was reading zeros.
Trying to rebuild now, but getting nothing on the serial.

finneyj, 2020-11-09:
Phone flash should definitely do it! it tested through fine on my build though.

I did rebuild your test though, maybe there's something else I missed...

finneyj, 2020-11-09:
uBit.init() ???

finneyj, 2020-11-09:
Above test works against master both with and without uBit.init() for me @microbit-carlos. 

Puzzling!

microbit-carlos, 2020-11-09:
The serial issue was a problem with my current setup (had tested something different before and left DAPLink in a bad state), so that's no problem.

I was able to test a rebuilt with the latest master in all the libraries and now works!
If the only change was the initial value I'm not sure why it wasn't working before, but we can consider this fixed 👍

finneyj, 2020-11-09:
hurrah!

Thanks @microbit-carlos,

closed 2020-11-09 by finneyj

microbit-carlos, 2020-11-09:
> Seems to be different behaviour on v1 and v2 for light level. On v2 the initial reading is 0 and on v1 still 255 as documented in
> [lancaster-university/microbit-dal#426](https://github.com/lancaster-university/microbit-dal/issues/426)

@microbit-mark have you tested this recently? If this is still the case we should open a new issue here.

A related issue I remember was found with the Scratch plugin, but I think ultimately the solution was to have an event to indicate when the light readings are ready? As the averaging wasn't good until enough readings were collected.

If the initial values are different in v1 and v2 we might want to look into that.

martinwork, 2020-11-09:
I have tested light level recently - both MakeCode and CPP. I get good readings from v1 and v2. v2 CODAL has a pause now, so doesn't deliver a bad first value. v1 DAL still starts at 255.

microbit-mark, 2020-11-10:
Thanks @martinwork. It appears this is fixed for codal, but still present in DAL as per lancaster-university/microbit-dal#426
@finneyj I don't think it was fixed in the DAL as it's still reproducing for David in MakeCode https://github.com/microsoft/pxt-microbit/issues/3451

microbit-carlos, 2020-11-10:
@finneyj if the original fix for this in the v1 DAL was to create an event to indicate when the readings are ready, do we have the same event in CODAL?

Was there a reason not to implement a fix like this one in v1 DAL? https://github.com/lancaster-university/codal-microbit-v2/commit/77693b2b283b7e456371c012f4c81b7017b1a58d#diff-64163a3b89cf79ee79d60156c44f4e855136b12a4130d6fdd7251c50b9bd0927R336

mod7ex, 2026-02-07:
Hi @martinwork thanks
this is the example https://makecode.microbit.org/_PvpVe54mt4hc it always shows 0 in complete dark or in sun light 
only when i cover the upper 5 LEDs it shows larger than 0