lancaster-university/codal-microbit-v2 #15: pulseIn implemented with CODAL

pelikhan, 2020-11-06:
The current pulseIn is not precise as it is based on pulling the pin sensor. Is it possible to implement "pulse" feature using CODAL pin events? Used for sonars and such.

finneyj, 2020-11-09:
Hi @pelikhan 

Yep - there's already an event based  implementation in there that is interrupt driven. e.g. someting like this works a charm for detecting the length of a HI pulse:

```
void
onPulse(Event e)
{
    DMESG("P0 PULSE: %d", (uint32_t) e.timestamp);
}

int 
main()
{
    uBit.init();

    uBit.io.P0.setPull(PullMode::Down);
    uBit.io.P0.eventOn(DEVICE_PIN_EVENT_ON_PULSE);
    uBit.messageBus.listen(uBit.io.P0.id, DEVICE_PIN_EVT_PULSE_HI, onPulse);
 
    while(1)
        uBit.sleep(10000);
}
```

You can use DEVICE_PIN_EVT_PULSE_LO if you want the opposite polarity. The ```uBit.io.P0.setPull(PullMode::Down)``` is purely for testing (you can try the above using some croc clips between P0 and 3V).

Will this do, or would you also like it packaged as a function that blocks the fiber until a pulse is detected, then returns the result? 

p.s. measurements are in microseconds, and the the timestamp field in event is a uint64_t on most platforms, as defined by CODAL_TIMESTAMP.

pelikhan, 2020-11-09:
Would be nice to have a helper. It is quite a common scenario to read a pulse in a loop for “sonar” apps.

pelikhan, 2020-11-09:
On the other hand, I think I can implement in the MaleCode side so that you don’t have to port it to the DAL.

finneyj, 2020-11-09:
We have the same API on both microbit-dal and codal.... so not a huge deal to port over.

Happy to package codal side if it helps - let me know my friend.

pelikhan, 2020-11-09:
Is it possible to wait for an event with a timeout?

finneyj, 2020-11-09:
not at present - sorry. Would be a nice feature though...

finneyj, 2020-11-09:
Hi @pelikhan 

See commit above for block pulseIn() behaviour as requested. Can you work with this here?   It is pretty generic, so could upstream into codal-core in due course.

example usage (delay and DMESG are just for diagnostics):

```
int 
main()
{
    uBit.init();
    uBit.io.P0.setPull(PullMode::Down);

    PulseIn *p = new PulseIn(uBit.io.P0);

    while(1)
    {
        int period = p->awaitPulse(5000000);
        DMESG("PULSE: %d uS", period);

        uBit.sleep(1000);
    }
}    
```

pelikhan, 2020-11-10:
Would be great to have this in CODAL definitly.

closed 2020-11-13 by pelikhan