lancaster-university/codal-microbit-v2 #18: Different V1/V2 behaviour of display.scroll when the display is disabled

martinwork, 2020-11-09:
If display.scroll is called while the screen is disabled V1 hangs whereas V2 continues.

In the test below, V2 sends  "release_fiber\n", V1 doesn't.

```
// This tests the behaviour of scroll when the display is disabled.

#include "MicroBit.h"

MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    //Called here is OK
    //uBit.serial.send( "scroll before disable\n");
    //uBit.display.scroll("0");

    uBit.serial.send( "disable\n");
    uBit.display.disable();

    //When called here, V1 hangs, V2 continues
    uBit.serial.send( "scroll after disable\n");
    uBit.display.scroll("0");

    uBit.serial.send( "release_fiber\n");
    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}
```

finneyj, 2020-11-09:
Thanks @martinwork.

Is that a bad thing? Feels more like a bug in v1 microbit-dal?

martinwork, 2020-11-09:
@finneyj Yes! Different rather than worse! I don't suppose anyone will be depending on the V1 behaviour, so long as the difference is limited to this circumstance.

finneyj, 2020-11-09:
Indeed.

martinwork, 2020-11-09:
Here's a different example. Probably a completely different issue, but I'm not sure what to call it!

V1 seems to run OK, but V2 appears to hang and never shows the count. It's OK without the scrollAsync in main. Maybe the idleCallback is doing something it shouldn't! 

```
#include "MicroBit.h"

#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#define target_panic(n) microbit_panic(n)
#define target_wait(n) wait_ms(n)
#endif

MicroBit uBit;

volatile int tick_count = 0;

class Component : MicroBitComponent
{
public:

    Component()
    {
      fiber_add_idle_component( this);
    }

    ~Component()
    {
    }

#if MICROBIT_CODAL
    void idleCallback()
#else
    void idleTick()
#endif
    {
      tick_count++;
      ManagedString s( tick_count % 10);
      uBit.display.printChar( *s.toCharArray());
      target_wait( 200);
    }
};

int
main()
{
    uBit.init();

    uBit.display.scrollAsync(".");

    new Component();

    release_fiber();
}

```

finneyj, 2020-11-09:
oooh - that's a sin. I never realized you had such an evil streak in you @martinwork. ;)

I think's it's because printChar() is a blocking call... so we shou;ldn't really be calling that from an idle call back (which is really not like running in a fiber). I think this is the source of the issue. I notice this seems to work, which is a non-blocking variant:

```
    {
      tick_count++;

      ManagedString s(tick_count % 10);
      uBit.display.image.print(*s.toCharArray());

      target_wait(200);
    }
```

I'm impressed v1 survived! 

Looking at the code, I not sure see why though, as the implementation is very similar:
https://github.com/lancaster-university/microbit-dal/blob/js-event-semantics/source/drivers/MicroBitDisplay.cpp#L480

closed 2021-06-07 by martinwork