lancaster-university/codal-microbit-v2 #20: Different I2C method signatures to the DAL drivers

microbit-carlos, 2020-11-09:
The I2C (and possibly other drivers?) in the micro:bit DAL use char:

https://github.com/lancaster-university/microbit-dal/blob/9164922a519aa213a565cbf2abaa2cf4debf4977/inc/drivers/MicroBitI2C.h#L104

And CODAL uses uint8_t:

https://github.com/lancaster-university/codal-nrf52/blob/master/inc/NRF52I2C.h#L73

So C++ code for v1 DAL might need to add casts to get it to compile for CODAL.

finneyj, 2020-11-09:
Thanks @microbit-carlos. 

I can't really change the CODAL interfaces at this point, but if we feel significant we could wrap the functions in the compat layer?

microbit-carlos, 2020-11-09:
> we could wrap the functions in the compat layer?

Yeah, what I was thinking that might be the only way to do this.
I don't think we've had anybody mentioning this before, @microbit-sam did this pop up at all in any of the MakeCode extensions? (with this or any other driver)

microbit-sam, 2020-11-09:
Ah yeah, not very common but there is a couple in the build logs (potentially more as the compiler stops at the first error)

```
error: no matching function for call to 'codal::NRF52I2C::NRF52I2C(PinName, PinName)'
 MicroBitI2C i2c(I2C_SDA0, I2C_SCL0);
```

finneyj, 2020-11-09:
i could just add this into the core I2C interface for backward compat... it's pretty harmless:

```
+
+    int I2C::write(int address, char *data, int len, bool repeated)
+    {
+        return write((uint16_t)address, (uint8_t *)data, len, repeated);
+    }
+
+    int I2C::read(int address, char *data, int len, bool repeated)
+    {
+        return read((uint16_t)address, (uint8_t *)data, len, repeated);
+    }
```

finneyj, 2020-11-09:
@microbit-sam  That error is more about the constructor though... which would imply a wrapper approach would be better. Where possible I like to surface the codal class rather than the compat layer classes, just so we don't get stuck in the past....

finneyj, 2020-11-27:
We've now added v1 compatible constructor wrappers and overloads for the char based I2C read/write methods, so I'll close this off. Please reopen if there are any issues with those solutions.

closed 2020-11-27 by finneyj