lancaster-university/codal-microbit-v2 #21: display.enable() doesn't re-configure the display pins

microbit-carlos, 2020-11-09:
With this example the rightmost column (pin10) is turned on while printing the text:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.display.disable();
    while (1) {
        uBit.sleep(250);
        uBit.io.P10.setAnalogValue(0);
        uBit.display.enable();
        uBit.display.print("Hello");
        uBit.display.disable();
    }
}
```

[display.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/5512511/display.hex.zip)

finneyj, 2020-11-09:
Thanks @microbit-carlos - fixed in master. 

Will be in next release (v0.2.17)

finneyj, 2020-11-13:
Works ok in my tests now @microbit-carlos. Please reopen if issue persists.

closed 2020-11-13 by finneyj