lancaster-university/codal-microbit-v2 #22: All CODAL related repos should to be tagged to formalise releases

microbit-carlos, 2020-11-10:
Currently untagged repos:
- https://github.com/lancaster-university/codal-nrf52
- https://github.com/lancaster-university/codal-core
    - This one is tagged, but the last tag was in Dec 2018 and I doubt we are using that version
- https://github.com/microbit-foundation/codal-microbit-nrf5sdk

microbit-carlos, 2020-11-10:
Should his file be pointing to tags as well? https://github.com/lancaster-university/codal-microbit-v2/blob/master/target.json

jamesadevine, 2020-11-10:
In CODAL tags are not created on libraries. Specific commit hashes are saved in the `target-locked.json` file in the target. `target.json` is used for development and checks out "development" branches (e.g. master/main) and should not have specific tags listed against each repository.

jamesadevine, 2020-11-10:
You could easily replace specific commit hashes with specific tags if required, but this will increase release time :smile:

microbit-carlos, 2020-11-10:
How is target-locked.json created and how does it work?
What is the clone-lock-build-archive process to ensure replicable builds?

microbit-carlos, 2020-11-10:
So I guess each new tag in this repo has an updated target-locked.json?
Does `python build.py` clone the repos from target-locked.json instead of target.json? Or is there an specific argument we need to ensure we are building from the locked commits?

jamesadevine, 2020-11-10:
> How is target-locked.json created and how does it work?

Creating the first `target-locked.json` is unfortunately a manual process. It is initially a copy of `target.json` and should be committed manually to master with a commit message of "Snapshot VX.X.X" (with X's replaced with the correct version number). 

After the initial creation of `target-locked.json`, the process becomes automatic. Passing the -l option to build.py causes it to iterate over each library entry in the `target-locked.json`, fetch its current commit hash, and store it in `target-locked.json`. Using -l will fetch the previous version number from git history and then auto increment the patch version. You can increment different parts of the semantic version by using -M (for major version) and -m (for minor version) in conjuction with -l. You can type `python build.py -h` to read this too!

>What is the clone-lock-build-archive process to ensure replicable builds?

When using target-locked.json, it will fetch each library in turn and check out that library at the git commit specified in json. You can also use any other git identifier to reference specific commits such as tags or branches.

> So I guess each new tag in this repo has an updated target-locked.json?

Correct.

> Does python build.py clone the repos from target-locked.json instead of target.json? Or is there an specific argument we need to ensure we are building from the locked commits?

Because the primary purpose of the build system is to provide reproducible binaries for MakeCode, by default the build system will use `target-locked.json` to check out the repositories. If you pass the -d flag (development) to build.py before any libraries have been cloned, it will check them out using `target.json`.

microbit-carlos, 2020-11-12:
That's really useful, thanks James!
We might still want to create some tags for major milestones, but storing the commit hashes on target-lock.json should be fine as well.

jamesadevine, 2020-11-12:
No problem! Yes, i'm sure CODAL could also be extended to automate that process. We ultimately just use python to echo git commands.

finneyj, 2020-11-13:
Thanks @jamesadevine.

I'm closing this off as I assume we're now happy with status quo, seeing as we can always extract the full set of dependencies for a given tag using build.py. 

Please reopen if you feel there's more to do/discuss @microbit-carlos

closed 2020-11-13 by finneyj