lancaster-university/codal-microbit-v2 #26: Button long press rarely indicated in Bluetooth notifications

bluetooth-mdw, 2020-11-16:
The Bluetooth LE button service has two characteristics, one representing the state of button A and the other, Button B. A value of 0 means not pressed, 1 means pressed and 2 means "long press". 

See https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html 

The way long press works on micro:bit V1, you press a button and this generates a notification with value==1 and keep holding for at least 2 seconds and this causes a second notification to be sent, this time with a value of 2.

In around 20 tests, I only saw a value of 2 (long press) transmitted in the button's characteristic value notification once. The other times, despite holding the button down for over 2 seconds, I only ever saw a single notification, with a value of 1.

@microbit-mark FYI

martinwork, 2020-12-12:
Thanks @bluetooth-mdw I have created https://github.com/lancaster-university/codal-microbit-v2/issues/53

martinwork, 2021-06-07:
@bluetooth-mdw The fix to https://github.com/lancaster-university/codal-microbit-v2/issues/53 was merged in v0.2.25.

bluetooth-mdw, 2021-06-07:
@martinwork Thanks for letting me know :-)

martinwork, 2021-06-07:
Sorry @bluetooth-mdw, a bit late, I know. I was checking through issues to tidy up... I can't close this. Would you like to close and reopen if it proves to be a problem still?

bluetooth-mdw, 2021-06-07:
Is v0.2.25 in use through makecode beta? I've never known how to check what version of the microbit C/C++ framework is in use from makecode. If it is, this is a very easy change for me to test.

martinwork, 2021-06-07:
`Is v0.2.25 in use through makecode beta?
`

MakeCode beta is now on v0.2.28. Live is v0.2.25.

`I've never known how to check what version of the microbit C/C++ framework is in use from makecode.
`

Me neither, until recently when someone made a video of the process...

MakeCode beta has both DAL and CODAL versions in the About dialogue now.

For older versions... Click the "makecode.microbit.org version" in the About dialogue, then the github tag short commit number, then Browse files, open pxtarget.json, and find "codalTarget".

bluetooth-mdw, 2021-06-07:
Thanks, that's a very handy tip!

And... I just tested and can confirm that this is indeed fixed so I shall close.

All the best, Martin.

closed 2021-06-07 by bluetooth-mdw