lancaster-university/codal-microbit-v2 #27: Bluetooth IO Pin Service not allowing control of pin connected devices

bluetooth-mdw, 2020-11-16:
The Pin Data characteristic belonging to the IO Pin Service allows the state of micro:bit pins to be changed, by writing to the characteristic value. How the data is interpreted depends on the value of other configuration characteristics in the same service. In the simplest case, Pin Data acts as a series of binary pin number/value pairs. 

Writing a value of 0x000101000200 to this characteristic should switch pin 0 on, pin 1 off and pin 2 off. Doing this using the nRF Connect application against a V1 micro:bit, with an LED connected to pin 0, works as expected. The LED switches on. Performing the same test on a micro:bit V2 does not work.

Use this code for testing: https://github.com/bittysoftware/bitty_controller_device_code/blob/master/microbit/makecode/switches.js 

FYI @microbit-mark

microbit-mark, 2020-11-16:
@martinwork is this fixed by https://github.com/microbit-foundation/codal-microbit/pull/132 and if so should we port this from the private to public repo?

martinwork, 2020-11-16:
@microbit-mark I think that PR should fix it. The order of the pins in uBit.io changed. I'll make another PR over there, if that's appropriate.

finneyj, 2020-11-16:
Hi @martinwork  @bluetooth-mdw 

I thought we already reordered the pins to accommodate:
https://github.com/lancaster-university/microbit-dal/blob/master/inc/drivers/MicroBitIO.h
https://github.com/lancaster-university/codal-microbit-v2/blob/master/model/MicroBitIO.h

All the edge connector pins seem to be in the same order?

p.s. Really nice to hear from you again @bluetooth-mdw! Been a long time!

martinwork, 2020-11-16:
@finneyj You are right, but I think I wrote this before they were reordered:
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/bluetooth/MicroBitIOPinService.cpp#L88

To be honest, I don't see how that would have worked then! It looks like I might have not finished it, and created the issue about changing the order instead!

finneyj, 2020-11-16:
Indeed, maybe the issue was a blocker and then finishing it just fell off the plate with everything else going on?

Do you fancy taking a look @martinwork ?

bluetooth-mdw, 2020-11-16:
Hello yourself @finneyj or should I say Professor? :-)

We should catch up!

finneyj, 2020-11-16:
haha - so you spotted that did you?  You always were a sharp one @bluetooth-mdw. ;)   

Yep, catch up sometime soon would be good!

martinwork, 2020-12-12:
@bluetooth-mdw This is fixed in v0.2.23. Tested with a hex from MakeCode beta and the micro:bit iOS app.

martinwork, 2021-04-19:
@bluetooth-mdw Please close if this is OK now.

bluetooth-mdw, 2021-04-19:
@martinwork Sorry,  I don't have time to test but I'll close all the same.

closed 2021-04-19 by bluetooth-mdw