lancaster-university/codal-microbit-v2 #28: Bluetooth all services sample hex - magnetometer service not working

bluetooth-mdw, 2020-11-17:
There's a link to a hex file which implements "all Bluetooth services" on this page: https://tech.microbit.org/latest-revision/ 

The magnetometer services does not work (no compass bearing data is transmitted when notifications are enabled). I think I wrote this sample originally and my guess is that it was not updated when work was done way back to change the accelerometer used and some of the rules for using it. Originally, use of the magnetometer service automatically triggered the calibration service. That's no longer the case afaik and so this sample needs to be changed to make an explicit call to calibrate the magnetometer or something else, equally cunning.

FYI @microbit-mark

martinwork, 2022-12-29:
The Bluetooth examples are now at https://tech.microbit.org/bluetooth/

@bluetooth-mdw 's suggestion to add an explicit call to calibrate the compass seems good to me. 

If I understand [this information](https://github.com/microsoft/pxt-microbit/issues/1750) correctly, it would be best for the samples to calibrate the compass, even if headings are not required.

I've created PR https://github.com/lancaster-university/microbit-v2-samples/pull/56.

Auto calibration was dropped in V1, and never happened in CODAL. Triggering calibration when the BLE bearing characteristic is read or subscribed to seems likely to be confusing.

As far as I can see, the DAL and CODAL `MicroBitMagnetometerService` always only updated the compass heading if the compass was calibrated, and didn't directly trigger a calibration, other than through the calibration characteristic.
 
In V1, the compass used to be calibrated automatically (if not already calibrated), when a compass related event listener was created, which happens in V1 when the `MicroBitMagnetometerService` is created. In V2 the listeners are added only while Bluetooth is connected.

V1 auto-calibration was prevented in @finneyj 's commit [here](https://github.com/lancaster-university/microbit/commit/d54e26fd4d2c0812923f60874ea17df24432f636), with CODAL created to match [here](https://github.com/lancaster-university/codal-microbit-v2/commit/46b7da26543c9bb66946eb39462ee7abeb6dd51e).

martinwork, 2023-01-18:
Just making a note for myself... The link to the hex is now at https://tech.microbit.org/bluetooth/

martinwork, 2023-01-18:
Created https://github.com/microbit-foundation/dev-docs/pull/391 with the new version in https://github.com/lancaster-university/microbit-v2-samples/pull/56 built with CODAL 0.2.40.

microbit-carlos, 2023-01-27:
PR https://github.com/microbit-foundation/dev-docs/pull/391 has been merged 🎉 
Thanks for the report and the PR!

closed 2023-01-27 by microbit-carlos