lancaster-university/codal-microbit-v2 #30: CODAL Sleep power consumption display oddity

jaustin, 2020-11-17:
Currently the sleep power is slightly higher than it needs to be because of the way the display sleeps.

Putting in a ticket to track sleep power consumption in CODAL.

I think the most likely is that we're not parking the display pins 'disconnected' appropriately in all cases.

finneyj, 2020-11-17:
Thanks @jaustin.

I'll take a look next time I find some time.

microbit-carlos, 2022-01-26:
Possible suspect: https://infocenter.nordicsemi.com/index.jsp?topic=%2Ferrata_nRF52833_Rev1%2FERR%2FnRF52833%2FRev1%2Flatest%2Fanomaly_833_246.html

## [246] System: Intermittent extra current consumption when going to sleep
This anomaly applies to IC Rev. Revision 1, build codes QIAA-Ax0, CJAA-Ax0, QDAA-Ax0.

It was inherited from the previous IC revision Engineering A.
### Symptoms
Extra current consumption in the range of 350 µA when in System On Idle.

### Conditions
A high-speed peripheral (CPU for example) accesses a RAM block which is being accessed by a low-speed peripheral through the DMA bus, with a specific timing, and the high-speed peripheral has higher priority than the low-speed peripheral.

### Consequences
Extra current consumption in System On Idle.

### Workaround
Apply the following code after any reset:
```c
*(volatile uint32_t *)0x4007AC84ul = 0x00000002ul;
```
Workaround consequences: Up to 40 µA current increase when the 16 MHz clock is used.
Parent topic: New and inherited anomalies

microbit-carlos, 2023-08-10:
## Extra power consumption cause

I've looked into this and found that commenting out this loop reduces the power consumption to around the expected levels:
https://github.com/lancaster-university/codal-microbit-v2/blob/4f2aedcd8e093e7ca9eefd574d4a647884e82505/source/NRF52LedMatrix.cpp#L194-L196

## Why? 🤷‍♂️

The additional power consumption is triggered when the `MODE` (disabled) or the `PSEL` (GPIO number) fields from `GPIOTE.CONFIG[n]` registers are cleared.

[image]

I've also noticed that from the 5 channels used (one per column), not all of them have the same impact to the additional power consumption triggered.

And I've also tried these different things without any noticeable improvement:
- first disabling all other GPIOTE & PPI config before clearing `GPIOTE.CONFIG`
- Clearing all GPIOTE events and latches
- Commenting out the sleep procedures from the other CODAL components configuring pins

## Power measurements

### Before (current state)

The power consumption slowly increases for around 10 seconds or so, this capture about a minute after system off show the stable average power consumption, around 900 uA @ 3V:
[image]

Increasing the voltage to 3.3V significantly decreases the frequency of the power peaks, and so the average current drops to 210 uA @ 3V3:
[image]

Of course, we would expect most batteries to be around the 2V5 (just pulled that number out of thin air, but it doesn't matter too much), and at that voltage the current consumption is a bit worse with more peaks but at a lower max, so we stay at around ~950 uA @ 2V5.
[image]

#### Bonus measurement: Current CODAL in DK board

This one is also a bit weird, loading the current CODAL hex into a DK board and measuring its power shows the current consumption slowly increasing over about 1 minute, up to ~ 300 uA @ 3V3 (it stabilised at around that point).
[image]

### micro:bit with workaround

~ 250 uA @ 2V5
[image]

~ 38 uA @ 3V

[image]

~ 45 uA @ 3V3
[image]

microbit-carlos, 2023-09-15:
Closing this as resolved in v0.2.59.

closed 2023-09-15 by microbit-carlos