lancaster-university/codal-microbit-v2 #41: Pulsed events on V2 Hardware

markb139, 2020-11-26:
Wasn't sure where to raise this. I've added a note to 
https://github.com/microsoft/pxt-microbit/issues/3731
However, on my V2 Microbit the pulse events aren't working. For examples
```javascript
control.onEvent(EventBusSource.MICROBIT_ID_IO_P0, EventBusValue.MICROBIT_PIN_EVT_PULSE_HI, function () {
    led.toggle(0, 2)
})
control.onEvent(EventBusSource.MICROBIT_ID_IO_P0, EventBusValue.MICROBIT_PIN_EVT_PULSE_LO, function () {
    led.toggle(1, 2)
})
pins.setEvents(DigitalPin.P0, PinEventType.Pulse)
basic.forever(function () {
    if (pins.digitalReadPin(DigitalPin.P0) == 1) {
        led.plot(0, 0)
    } else {
        led.unplot(0, 0)
    }
})
basic.forever(function () {
    pins.digitalWritePin(DigitalPin.P1, 0)
    basic.pause(500)
    pins.digitalWritePin(DigitalPin.P1, 1)
    basic.pause(500)
})
```
The code works as expected on V1 hardware. Note, that if the event type is changed to edge detection, then that works fine on V1 & V2. So it looks like just the pulse is not working

finneyj, 2020-11-26:
Hi @markb139 

Thanks for the bug report. This looks like a bug either here in codal or in the MakeCode layer that consumes the codal APIs... so quite possibly the right place to raise this issue. 

Can you please describe the behaviour that you see? Do you receive any events at all? 

I wonder if this is related to the fact that you are measuring both HI and LO pulses at the same time? Does it make any difference if you register for a single PULSE event (either HI or LO) ?

markb139, 2020-11-26:
On the V2 microbit I don't receive the onEvent for either HI or LO. On my V1 microbit the LEDs flash after the pulses.
I just tried just the HI and then just the LO onEvent handlers. Still not working on the V2 microbit

finneyj, 2020-11-26:
Thanks @markb139.

Yes, looks like an initialization bug crept in when the NRF52Pin::getPulseUs() method was introduced recently. Fixed in here:
https://github.com/lancaster-university/codal-nrf52/commit/6a8f1d5effdef4a177b6e8cbed91042a619bcd64

This code now works as expected C++ side:

```
void onHi(MicroBitEvent e)
{
    DMESG("P0: PULSE HIGH");
}
void onLo(MicroBitEvent e)
{
    DMESG("P0: PULSE LO");
}

void
pulse_event_test()
{
    uBit.io.P1.setPull(PullMode::Up);
    uBit.io.P1.eventOn(DEVICE_PIN_EVENT_ON_PULSE);

    uBit.messageBus.listen(uBit.io.P1.id, DEVICE_PIN_EVT_PULSE_HI, onHi);
    uBit.messageBus.listen(uBit.io.P1.id, DEVICE_PIN_EVT_PULSE_LO, onLo);

    while(1)
        uBit.sleep(1000);
}
```

This fix should work its way through in the next update.

markb139, 2020-11-27:
The C++ version of my code works now. Thanks

finneyj, 2020-11-27:
cool - thanks for confirming.

closed 2020-11-27 by finneyj