lancaster-university/codal-microbit-v2 #42: Difference in note/melody behaviour between DAL and CODAL

microbit-mark, 2020-12-01:
In DAL notes of the same pitch and length had some definition or slight pause between the notes played, so you could distinguish by ear that the same pitch was being played twice. In CODAL this is no longer the case.

Using a pre-built melody such as Ode To Joy demonstrates this.

[image]

1. Flash this program from https://makecode.microbit.org/#editor
2. Flash this program from https://makecode.microbit.org/v2/#editor

You will hear the audible difference.

This also reproduces in MicroPython 2.0.0-beta.1 vs MicroPython 1.0.1

martinwork, 2021-06-07:
Samples: old makecode (v1 only), current makecode, current python... [ode_hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6608928/ode_hex.zip)

In current MakeCode, V1 sounds much like V2 to me - the separation between notes is missing compared to a hex compiled with the older makecode version. V1 maybe has a slight something between some notes, but it sounds more like a glitch than the previous pause.

In current MicroPython (tested with beta), there is a pause between notes, but it sounds exaggerated between same-pitch notes.

The MakeCode function analogPitch(int frequency, int ms) used to have a wait_ms(5), then fiber_sleep(5).
The pause was commented out last November:
https://github.com/microsoft/pxt-microbit/blob/d80514bda223f729d452c1e51047f3272a2317fb/libs/core/pins.cpp#L384
in this commit:
https://github.com/microsoft/pxt-microbit/commit/249873ed474aef27f489fff7897297c8e5906806

That commit mentions "joe" (@finneyj ?), and the code comment says "causes issues with v2 DMA"

finneyj, 2021-06-07:
The issue here is that v1 had an "instantaneous" audio implementation. It could only do raw, unmodulated square wave outputs. So if you stopped playing, paused the CPU and started playing, that pause would defined the "quiet" time between notes.

v2 however, is a buffered, DMA based PCM approach (much more like the soundcard on a PC). Slowing the CPU here only controls the gap between buffers being queued. Which can have a much worse effect on playout quality. 

This could be addressed simply enough by actively injecting silence into the audio stream. However, we are hampered by the constraint that the purpose of this API is to be backward compatible with v1. So the currently behaviour was chosen as the best that was achievable under those constraints.

martinwork, 2021-06-07:
The fiber_sleep(5) pauses become possible with "CONFIG_SOUND_OUTPUT_PIN_PERIOD": 5 in codal.json. Also a passable sonic screwdriver, as in https://github.com/microbit-foundation/micropython-microbit-v2/issues/67. I guess this isn't a good idea for the default.

Press A or B to try: [test-melody.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6612205/test-melody.zip)

martinwork, 2021-06-08:
I added a global count of the number of buffers and bytes returned by SoundEmojiSynthesizer::pull().
10 x 300ms notes and 5ms pauses typically take 3080-3093ms for 533-535 buffers containing 272896-273920 bytes.
Here are the values I recorded:

```
5ms
3082 533 272896
3089 534 273408
3082 534 273408
3089 534 273408
3089 535 273920
3093 535 273920
3089 534 273408
avg 3,087.6
 
50ms
3082 533 272896
3086 534 273408
3093 535 273920
3093 535 273920
3090 535 273920

reset
3081 533 272896
3092 534 273408
3088 534 273408
3088 534 273408
3085 534 273408
3089 535 273920
3088 534 273408
avg 3,087.3

reset
3080 533 272896
3092 535 273920
3092 535 273920
3089 535 273920

```

dpgeorge, 2021-08-09:
@finneyj is there any downside to setting `CONFIG_SOUND_OUTPUT_PIN_PERIOD` to 5 (ms)?  Will this affect sound emoji output/rendering?

finneyj, 2021-08-09:
Hi @dpgeorge 

It wouldn't affect any sound emoji behaviour I don't think. Sound emojis are synthesized through a different API to this.

I don't think there would be any implications from doing that, apart from the fact that 5ms is very short period for a note... so could cause a bit more CPU churn, but probably not a huge amount more.

Generally though, approaches to injecting silence based on delaying the CPU will always be inaccurate when we're using DMA. I guess the question is whether increasing the update frequency here by a factor of ten makes it "good enough" to warrant the additional CPU overhead?

dpgeorge, 2021-08-10:
> It wouldn't affect any sound emoji behaviour I don't think. Sound emojis are synthesized through a different API to this

Great!

> I don't think there would be any implications from doing that, apart from the fact that 5ms is very short period for a note... so could cause a bit more CPU churn, but probably not a huge amount more.

It's not the length of the note but rather the granularity/quanta of lengths of notes and lengths of gaps.

I think 50ms is too long compared to what human ears notice.  In MicroPython's music module there is a 10ms break between notes, and it's noticeable if this is increased to a 50ms break.

> Generally though, approaches to injecting silence based on delaying the CPU will always be inaccurate when we're using DMA. I guess the question is whether increasing the update frequency here by a factor of ten makes it "good enough" to warrant the additional CPU overhead?

It's not really injecting silence, it's playing a note for a given period of time.  And when the note is turned off it's quite unexpected that it should take between 0ms and 50ms to actually turn off.  Even if silence was injected into the DMA stream to make it accurate, durations are still quantised to 50ms.

Consider the following example:
```c
int main() {
    uBit.audio.virtualOutputPin.setAnalogPeriodUs(2000);
    for (int i = 0; i < 20; ++i) {
        uBit.audio.virtualOutputPin.setAnalogValue(128);
        uBit.sleep(40);
        uBit.audio.virtualOutputPin.setAnalogValue(0);
        uBit.sleep(40);
    }
}
```
This definitely does not make 20 even spaced beeps!  Each beep and each gap goes for 0, 50 or 100ms.

I guess the alternative to the above is to generate a full waveform with 20 beeps and feed that waveform into the mixer.  But that's pretty expensive in RAM usage or difficult to implement.  It's also not possible to take this approach if the user wants to make some fancy sound effect by turning on/off the virtualOutputPin quickly (eg the sonic screw driver example).

Setting `CONFIG_SOUND_OUTPUT_PIN_PERIOD` to 5 does solve most of these problems and I think is definitely worth the increased CPU consumption.

finneyj, 2021-08-10:
Hi @dpgeorge 

Yes - exactly right. I'm in total agreement. Apologies - what I was trying to say above is precisely what you describe here - that we're controlling time quantisation. Hence, we it can never be "perfect". All we can do through this API is an approximation...

If 5ms gives us a sufficiently improved resolution that it enable more use cases/better backward compatibility, I agree it is worth the additional COU cost.

finneyj, 2021-08-10:
Applied in https://github.com/lancaster-university/codal-microbit-v2/commit/6e880bc3d10df1a585bb9046ee4ab02bdcab1f34

dpgeorge, 2021-08-10:
Great, thank you!

microbit-carlos, 2022-05-05:
@JohnVidler could you give this a quick test before we close it? Thanks!

JohnVidler, 2022-05-05:
Tested on a v2, and it works :) Closing.