lancaster-university/codal-microbit-v2 #44: Getting error code 090 on new microbit

KalbeAbbas, 2020-12-04:
I am using SPI functions for writing to SD card with [new microbit](https://microbit.org/new-microbit/) and getting error code 090 on microbit screen every time after writing two lines to SD card as if its is periodic, writes two lines shows error code 090 and restarts again.

finneyj, 2020-12-04:
Hi @KalbeAbbas 

An 090 panic code is a hardware configuration error. You can see these list of panic codes here:
https://github.com/lancaster-university/codal-core/blob/master/inc/core/ErrorNo.h#L101

Looking at the SPI driver, it looks like you are trying to create the SPI driver multiple times:
https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52SPI.cpp#L55

Could you perhaps share some of your code so we can see how this might be happening?

KalbeAbbas, 2020-12-04:
Hi @finneyj, Thanks.

I first define function for allocating SPI pins:

```
static SPI* allocSPI() {
	SPI* spi = NULL;
        spi = new SPI(MOSI, MISO, SCK);
        return spi;
}

static SPI* p = NULL;
```

Then I allocate pins, set frequency and mode:

```
  CS_INIT();  // Chip Select Init
  CS_H();  // Chip Select High
  p = allocSPI();
  p->frequency(10000);
  p->format(8, 0);
```

and these are functions to write and read to/from SD card, they are called every time when writing:

```
static void xmit_mmc(
    const BYTE *buff,
    UINT bc)
{
  BYTE d;

  do
  {
    d = *buff++;
	p->write((int)d);
  } while (--bc);
}
```

```
static void rcvr_mmc(
    BYTE *buff,
    UINT bc)
{
  BYTE r;

  do
  {
    *buff++ = p->write(0xFF);
  } while (--bc);
}
```

These are low-level functions, all other function call these functions at lowest-level. Let me know if you want any other info

KalbeAbbas, 2020-12-04:
Forgot to mention, I call allocSPI() function to allocate SPI only once at the beginning.

finneyj, 2020-12-04:
That's curious.

I'm afraid I can't see any other way that panic code can be generated apart from calling the NRF52SPI constructor multiple times... All the other causes are raised from other modules:

```
libraries/codal-nrf52/source/NRF52I2C.cpp:        target_panic(DEVICE_HARDWARE_CONFIGURATION_ERROR);
libraries/codal-nrf52/source/NRF52Serial.cpp:        target_panic(DEVICE_HARDWARE_CONFIGURATION_ERROR);
libraries/codal-nrf52/source/NRF52SPI.cpp:        target_panic(DEVICE_HARDWARE_CONFIGURATION_ERROR);
libraries/codal-nrf52/source/peripheral_alloc.cpp:    target_panic(DEVICE_HARDWARE_CONFIGURATION_ERROR);
libraries/codal-nrf52/source/ZSingleWireSerial.cpp:        target_panic(DEVICE_HARDWARE_CONFIGURATION_ERROR);
```

finneyj, 2020-12-04:
Are you really, really sure there's no way the your initialization code is somehow being called twice? Maybe try putting some debug statement in there to be sure?

 ;)

KalbeAbbas, 2020-12-04:
Good idea. I'll try that :)

KalbeAbbas, 2020-12-04:
I guess you are right, its called each new cycle. Let me confirm if the same error code comes again.

KalbeAbbas, 2020-12-04:
@finneyj Thank you. That solved the issue :)

KalbeAbbas, 2020-12-04:
I am closing this issue now

closed 2020-12-04 by KalbeAbbas

finneyj, 2020-12-04:
Great! Thanks for reporting back @KalbeAbbas 

If it make you feel better I'm teaching an Computer Science class as we speak.... and you fixed your code faster than my students did. ;)