lancaster-university/codal-microbit-v2 #45: On power up, V2 sees garbage appear in uBit.serial.read buffer

martinwork, 2020-12-07:
Typically, after power up but not reset, a single random byte appears in the serial read buffer between uBit.systemTime()s 149 and 808.

On battery power, with no USB lead connected, the same thing happens, and sometimes I see 13 garbage bytes that take much longer appear.

[serial-startup-hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/5653844/serial-startup-hex.zip)
```
#include "MicroBit.h"

#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif

MicroBit uBit;

int serial_BAUDRATE = 115200;
//int serial_BAUDRATE = 57600;

MicroBitSerialMode serial_READMODE = ASYNC;
MicroBitSerialMode serial_SENDMODE = ASYNC;
//MicroBitSerialMode serial_SENDMODE = SYNC_SLEEP;

int serial_BUFFER_SIZE = ((uint8_t)250);

void serial_startup()
{
#if MICROBIT_CODAL
    uBit.serial.setBaudrate( serial_BAUDRATE);
#else
    uBit.serial.baud( serial_BAUDRATE);
#endif

    uBit.serial.setRxBufferSize( serial_BUFFER_SIZE);
    uBit.serial.setTxBufferSize( serial_BUFFER_SIZE);

    uint64_t t0 = uBit.systemTime();
    uint64_t t1 = 0;
    uint64_t t2 = 0;
    uint64_t tnow = 0;

    uint8_t store[1024];
    int count = 0;

    while ( tnow = uBit.systemTime(), tnow - t0 < 5000)
    {
        uint8_t buffer[serial_BUFFER_SIZE];
        int received = uBit.serial.read( buffer, serial_BUFFER_SIZE, serial_READMODE);
        if (received > 0)
        {
          if ( t1 == 0) t1 = tnow;
          t2 = tnow;

          if ( count + received < sizeof(store))
          {
            memcpy( store + count, buffer, received);
            count += received;
          }
        }
    }

    if ( count > 0)
    {
      uBit.display.scroll( ManagedString( count));
      uBit.display.scroll( ManagedString( (int)t1));
      uBit.display.scroll( ManagedString( (int)t2));

      uBit.serial.printf( "t0 %d t1 %d t2 %d count %d\n", (int)t0, (int)t1, (int)t2, count);

      for ( int i = 0; i < count; i++)
        uBit.serial.printf( "%x %d %c \n", (unsigned long) store[ i], (int) store[ i], (char) store[i]);
    }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    serial_startup();

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

martinwork, 2020-12-07:
A simpler example. It would appear that setRxBufferSize triggers the problem. Inserting a sleep(2000) before calling setRxBufferSize stops it.

```
#include "MicroBit.h"

#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif

MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    //uBit.sleep(2000);
    //if ( uBit.serial.isReadable())
    //  microbit_panic(888);

//#if MICROBIT_CODAL
//    uBit.serial.setBaudrate( 115200);
//#else
//    uBit.serial.baud( 115200);
//#endif

    uBit.serial.setRxBufferSize( 250);
    //uBit.serial.setTxBufferSize( 250);

    uBit.sleep(2000);
    if ( uBit.serial.isReadable())
      microbit_panic(999);

    while (true)
    {
      if ( uBit.serial.isReadable())
      {
          int c = uBit.serial.getc();
          if ( c >= 0)
              uBit.serial.putc( c);
          else
              uBit.serial.putc('?');
      }
      uBit.sleep(0);
    }

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

finneyj, 2020-12-07:
Interesting.  Resizing the buffer could certainly explain it - it's not somethin I do on most tests, and the UART receiver is DMA enabled.... so it could be that when the RX buffer is resized, the hardware is still DMA'ing into the old buffer until it is full... but as it reports data is received, some uninitialized data makes its way to the application. 

I'm not sure why that would behave differently on a cold start compared to a soft reset though. 

Let's decide what we want to do with #46 first, as that will likely influence how we fix this....

martinwork, 2020-12-08:
See https://github.com/microbit-foundation/micropython-microbit-v2/issues/40 for a python program that demonstrates this issue.

microbit-carlos, 2020-12-10:
Just as another data point, I tried removing the `uBit.serial.setRxBufferSize()` from MicroPython, replaced by a simple `uBit.serial.read(ASYNC)` to ensure things are initialised :

https://github.com/microbit-foundation/micropython-microbit-v2/blob/a76e1413bcd66f128a31d98756fc3d1f336d1580/src/codal_app/main.cpp#L61

And I still experience the same problem:
- [MICROBIT.hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/5671737/MICROBIT.hex.zip)

finneyj, 2020-12-16:
@martinwork @dpgeorge 

I think this should be resolved in this commit:
https://github.com/lancaster-university/codal-nrf52/commit/efd63319af7c5f8034a4be120cf9bd781ba7c3d2

martinwork, 2020-12-17:
Thanks @finneyj It looks good to me - no garbage!

finneyj, 2020-12-17:
Great. Let's close this one off then.

closed 2020-12-17 by finneyj

microbit-carlos, 2020-12-17:
@finneyj could you create a new tag to be able to update MicroPython and test over there?

finneyj, 2020-12-17:
Yep - Just wanted to squeeze in a patch for the I2C bus probe first.

v0.2.24