lancaster-university/codal-microbit-v2 #46: Serial RX buffer overflow behaviour

martinwork, 2020-12-07:
When pasting enough text (example below) into tera term to overflow the buffer, this serial echo program, returns characters that are out of order. Subsequent single, typed characters (e.g. space) are echoed with characters from the pasted text. Buffering doesn't fail with the Mbed example.

[serial-echo hex.zip](https://github.com/lancaster-university/codal-microbit-v2/files/5655405/serial-echo.hex.zip)

```
#include "MicroBit.h"

#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif

MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    //uBit.sleep(2000);
    //if ( uBit.serial.isReadable())
    //  microbit_panic(888);

#if MICROBIT_CODAL
    uBit.serial.setBaudrate( 115200);
#else
    uBit.serial.baud( 115200);
#endif

    uBit.serial.setRxBufferSize( 250);
    uBit.serial.setTxBufferSize( 250);

    uBit.sleep(2000);
    if ( uBit.serial.isReadable())
      microbit_panic(999);

    while (true)
    {
      if ( uBit.serial.isReadable())
      {
          int c = uBit.serial.getc();
          if ( c >= 0)
              uBit.serial.putc( c);
          else
              uBit.serial.putc('?');
      }
      uBit.sleep(0);
    }

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

```
123456789A
123456789B
123456789C
123456789D
123456789E
123456789F
123456789G
123456789H
123456789I
123456789J
123456789K
123456789L
123456789M
123456789N
123456789O
123456789P
123456789Q
123456789R
123456789S
123456789T
123456789U
123456789V
123456789W
123456789X
123456789Y
123456789Z

123456789A
123456789B
123456789C
123456789D
123456789E
123456789F
123456789G
123456789H
123456789I
123456789J
123456789K
123456789L
123456789M
123456789N
123456789O
123456789P
123456789Q
123456789R
123456789S
123456789T
123456789U
123456789V
123456789W
123456789X
123456789Y
123456789Z

```

```
#include "mbed.h"

DigitalOut rows[5] = {
    DigitalOut(ROW_1, 0),
    DigitalOut(ROW_2, 0),
    DigitalOut(ROW_3, 0),
    DigitalOut(ROW_4, 0),
    DigitalOut(ROW_5, 0)
};
DigitalOut cols[5] = {
    DigitalOut(COL_1, 1),
    DigitalOut(COL_2, 1),
    DigitalOut(COL_3, 1),
    DigitalOut(COL_4, 1),
    DigitalOut(COL_5, 1)
};

Serial pc(USBTX, USBRX);

int main() {
    for ( int i = 0; i < 5; i++)
    {
      rows[i] = 1;
      cols[i] = 1;
    }

    pc.baud(115200);

    while (1)
    {
        if ( pc.readable())
        {
            cols[0] = 0;
            pc.putc( pc.getc());
        }
    }

    return 0;
}

```

finneyj, 2020-12-07:
Thanks @martinwork.

Are the received bytes reordered, or just delayed? (sorry, but I'm not seeing any reordering in your example data trace above?)

finneyj, 2020-12-07:
I did observe some cases where a NRF_UARTE_EVENT_RXDRDY interrupt seems to gets "lost" when. I suspect this was due to a higher priority IRQ racing it. I did at the time do two things:

1) increase the priority of the serial interrupts such that this didn't happen
2) update the code so that it always caught up at the end of the DMA buffer (which is the only time that you can get feedback form the hardware on how many bytes have actually been received). 

So I am a bit curious about your IRQ configuration @martinwork. Are you perhaps running with SoftDevice active? and that could racing the serial IRQ? An IRQ set with a higher priority than the UART?

p.s. I took a quick look at the mbed driver, to see how that worked... and was quite entertained to see how the nordic engineers solved the problem:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L90

Would be interested to welcome feedback here on how we'd like to go forward, as I think we're looking at a hardware limitation.
I think our choices are:

1) We stuck with what we have, which is more efficient but might sometimes lag by one or two bytes until the DMA buffer fills.
2) We reduce the DMA buffer size to a single byte (as per mbed implementation), and rely on the hardware FIFO (which is just a handful of bytes long) to mask over the the data receive interrupt. 

Any thoughts/opinions?

martinwork, 2020-12-08:
I pasted the 0123456789A data as above into tera term, and see the echoes below, which goes wrong after 120 characters , after 123456789L, and stops with 1234567.

Then pressing SPACE repeatedly adds one of the originally pasted characters, so it looks like SPACE is being echoed with previous buffer contents, maybe?

Original echoes:
```
123456789A
123456789B
123456789C
123456789D
123456789E
123456789F
123456789G
123456789H
123456789I
123456789J
123456789K
123456789L
9I
423456789J
123456789K
123456789L
123456789M
123456789N
123456789O
123456789P
123456789Q
123456789R
123456789S
123456789T
123456789U
123456789V
123456789W
123456789X
123456789Y
123456789Z
3456789D
123456789E
123456789F
123456789G
123456789H
123456789I
123456789J
123456789K
123456789L
123456789M
123456789N
1234567

```

Then pressing SPACE repeatedly, echoes with other buffer contents:
```
89O
123456789P
123456789Q
123456789R
123456789S
123456789T
123456789U
123456789V
123456789W
1234
```
etc.

martinwork, 2020-12-08:
Notice there are 3 Js in the echo though only 2 were pasted.

finneyj, 2020-12-08:
Thanks @martinwork - most helpful. I think this could only be caused by the rung buffer getting out of sync (i.e. the driver filling up the wrong half of the buffer with DMA). Not sure how that could happen though. curious. 

I'll try to look at it this evening - i'm totally maxed out with the day job this week I'm afraid.

p.s. Are you running with SoftDevice enabled in your tests?

martinwork, 2020-12-08:
I am building master with a pairing mode only codal.json.

microbit-carlos, 2020-12-09:
@finneyj is there anything we can do to help debug this? Based on https://github.com/lancaster-university/codal-microbit-v2/issues/45#issuecomment-740230050 this issue is blocking https://github.com/lancaster-university/codal-microbit-v2/issues/45 and that one is affecting the Mu editor to access the MicroPython REPL to be able to flash the micro:bit.

finneyj, 2020-12-16:
@martinwork @microbit-carlos 

I've made some improvements in this patch that appear to resolve this issue in my tests:
https://github.com/lancaster-university/codal-nrf52/commit/efd63319af7c5f8034a4be120cf9bd781ba7c3d2

Also some more general performance and behavioural improvements in here:
https://github.com/lancaster-university/codal-core/commit/e6b73051ed273a288536b523227915d24be01b79

Can you please run some of your tests again on this revision @martinwork ?

martinwork, 2020-12-16:
Using the sample and test paste above, the echo in tera term is something like below. It varies a bit. It looks like all the variants have the right number of 123s.

```
123456789A
123456789B
123456789C
123456789D
123456789E
123456789F
123456789G
123456789H
123456789I
123456789J
123456789K
123456789L
123456789M
123456789N
123456789O
123456789P
123456789Q
123456789R
123456789S
123456789T
123456789U
123456789V
123456789W
123456789X
123456789Y
123456789Z

123456789A
123456789B
123456789C
123456789D
123456789E
123456789F
123456789G
123456789H
12345678912312341234123412341234123412312341234123412341234123412312341234
```

finneyj, 2020-12-16:
Thanks @martinwork.

is that.... a good thing? 

:)

martinwork, 2020-12-16:
Just an observation! To me it means bytes are being dropped, and what's getting through is in the right order. It's a bit odd that all the "123"s are there rather than being a bit more "random". If I repeat the paste several times, there seem to be a few patterns that keep cropping up. I don't see the previous very odd behaviour where space or return are echoed with other characters.

I'm not sure what we expect when we deliberately send bytes fast enough to overflow the buffer. I expect it to go badly!

finneyj, 2020-12-16:
Indeed. We can't avoid losing data when we have an overflow condition. As long as the data coming through remains in order, I think that's fine.

p.s. it's probably also worth noting here that putc() is a very low level call, so actually bypasses the transmission buffer and will go straight to the DMA buffer. If you use sendChar() instead, it will return as soon as the byte enters the transmission ringbuffer.... so should perform faster and will preserve total ordering w.r.t. other send() calls.

martinwork, 2020-12-17:
Thanks @finney, for the hint about putc(), which came from creating the mbed sample first. I've changed to read() and sendChar(c). The response to the test paste seems the same.

V1 and V2 now behave essentially the same in this test, as far as I can see, except V1 starts missing characters sooner.

The purpose of overflowing the buffer was to reveal the kind of strange behaviour seen in the REPL. I don't see any sign of that now in this test.

The right number of 123s suggests it's still correctly processing incoming characters in order to the end. The odd letter that doesn't get missed is in the right place among the 123s. Is newline looking after number 1?

Can we close this now?

finneyj, 2020-12-17:
Yep, I'm happy to close.

closed 2020-12-17 by finneyj