lancaster-university/codal-microbit-v2 #47: uBit.serial.isReadable() always returns false unless uBit.serial.setRxBufferSize() is called

martinwork, 2020-12-08:
The program below echoes characters typed into tera term on Windows 10, but if the setRxBufferSize() call is removed, isReadable() always returns false.

```
#include "MicroBit.h"

MicroBit uBit;

void serial_fiber()
{
    uint8_t chr;

    // without setting the buffer, isReadable() never returns true

    uBit.serial.setRxBufferSize( 250);

    while (true)
    {
      if ( uBit.serial.isReadable())
      {
        //uBit.display.print('R');
        chr = '?';
        uBit.serial.read( &chr, 1);
        uBit.serial.send( &chr, 1);
      }

      uBit.sleep(0);
    }
}

void togglePixel( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue( x, y) ? 0 : 255);
}

void forever()
{
    while (true)
    {
      togglePixel( 0, 0);
      uBit.sleep(1000);
    }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    create_fiber( forever);

    create_fiber( serial_fiber);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

dpgeorge, 2020-12-14:
Yes I also noticed this.  The reason is because `Serial::initialiseRx()` is not (lazily) called by `isReadable()`, only by the `read()` methods.

finneyj, 2020-12-16:
Yes indeed - thanks @dpgeorge.

Fixed in:
https://github.com/lancaster-university/codal-core/commit/e6b73051ed273a288536b523227915d24be01b79

martinwork, 2020-12-17:
Thanks @finneyj . Working now.

closed 2020-12-17 by martinwork