lancaster-university/codal-microbit-v2 #48: Is ADC read different on V1 and V2?

microbit-mark, 2020-12-08:
This MakeCode example produces different results on each board.

[image]

https://makecode.microbit.org/_ehJ13Hej17MW

V1 returns 131µs and V2 returns 246µs

Is this a difference in DAL/CODAL?

finneyj, 2020-12-08:
Hi @microbit-mark 

Kind of. More of a difference in how the v1 and v2 hardware behaves. Both devices on have a single ADC, but on v2 the ADC is also used to read the microphone, so the edge connector analogue read operations need to share that hardware. Not surprising that there's a longer startup time due to this.

You should find that subsequent reads are faster.

JLFra, 2020-12-08:
Hello,

Thank you. I have just tested, in fact, we fall back to the second acquisition at 130µs. But it does not seem normal to me that it is necessary to launch an acquisition for nothing, it is not very logical. It would be good if you change your firmware instead to allow us to always have the same acquisition time. Thanks for doing what's needed.
Your sincerely.

Julien

finneyj, 2020-12-08:
Hello @JLFra 

The acquisition time will be the same in all cases - it's simply that the first time a call is made it needs to set up the ADC DMA pipeline... 

If this is a problem for you, perhaps you could explain why?

JLFra, 2020-12-09:
Sometimes we need to keep the sampling time as low as possible, including at the beginning. You will easily understand that to sample a signal you have to know the sampling frequency to respect shannon. In short, it would be good if your system is reliable and this frequency does not vary according to different parameters. Could you arrange for your firmware to correctly set this frequency at startup. Thanks.

finneyj, 2020-12-18:
Hi @JLFra 

Indeed - I fully understand the need isochronous sampling for any applications that require high fidelity data. Harry Nyquist would also agree of course!

I'm afraid this isn't just a case of the frequency being set differently on the first call - the difference in latency you see is due to the time it takes to bring up and configure the ADC hardware on the NRF52 before the first sample is taken and reported. There's not really anything I can do about that for a number of reasons:

1) The ADC needs to be brought up on demand (like we do with all other micro:bit peripherals). If we bring it up at boot time, it would be constantly eating battery power and CPU resource for every application. Even those that don't use the ADC... and that's not really OK. 

2) Even if we did bring up the ADC at boot time, there would still be an additional latency anyway, caused by switching the ADC's sample interleaving pattern to account for whichever pin you select. At best, that would take up to one DMA buffer's worth of time to resolve.

3) The blocks program above is really the wrong API for gathering isochronous high frequency sample data. By running every sample through the CPU in this way, the program will be very much at the mercy of jitter caused by other demands on the CPUs time, e,g. whenever any interrupts fire (e.g. to service the display, serial port, sound, microphone, light sensing, pin events, scheduler, timers, radio...), you would see a much bigger spike in latency than that associated with bringing up the ADC hardware.

Now, the good news is that the v2 hardware and is capable of lossless, DMA enabled buffered streaming of sample data. This is already implemented in the underlying C++ (CODAL) layer that sits just below MakeCode. This is what we use to gather microphone data, and is also the API you are transparently using in the code above (just not in the most efficient way).

We're already in the process of defining a streamed data input pipeline, which could be surfaced in TypeScript and maybe even Blocks. e.g. imagine an event handler that delivers a buffer of samples - much like we do today for the RADIO package. This API isn't ready yet in MakeCode though. I think it would be possible to consume that C++ API directly via a custom MakeCode extension, should you wish to jump ahead before we surface a standardized API.

I hope this helps.