lancaster-university/codal-microbit-v2 #49: i2c.write hangs with a NULL buffer

martinwork, 2020-12-08:
micropython has a function i2c.scan, which hangs V2 when an external i2c device is connected.

As far as I understand the micropython source, this is implemented by writing a NULL buffer to each address.

Have I got it right @dpgeorge ? I'm puzzled because my sample doesn't detect devices on a V1, whereas i2c.scan does, so I must be doing something wrong!

V2 i2c.write hangs when it gets to an attached i2c device - maybe stuck in waitForStop.

With the sample below, Press A to write to each address; B to read from each address.

```
#include "MicroBit.h"

#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif

MicroBit uBit;

void i2c_scan_write()
{
    for (int addr = 0x08; addr < 0x78; ++addr)
    {
        int ret = uBit.i2c.write( addr << 1, NULL, 0, false);

        uBit.serial.printf( "write addr %x = %d\n", addr << 1, ret);
    }
}

void i2c_scan_read()
{
#if MICROBIT_CODAL
    uint8_t data[100];
#else
    char data[100];
#endif

    for ( int addr = 0x08; addr < 0x78; addr++)
    {
        memset( data, 255, 100);
        int ret = uBit.i2c.read( addr << 1, data, 1, false);
        uBit.serial.printf( "read addr %x = %d data[0] = %d\n", addr << 1, ret, (int) data[0]);
    }
}

void onButtonA(MicroBitEvent e)
{
  i2c_scan_write();
}

void onButtonB(MicroBitEvent e)
{
  i2c_scan_read();
}

void togglePixel( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue( x, y) ? 0 : 255);
}

void forever()
{
    while (true)
    {
      togglePixel( 0, 0);
      uBit.sleep(1000);
    }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    create_fiber( forever);

    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

dpgeorge, 2020-12-10:
> As far as I understand the micropython source, this is implemented by writing a NULL buffer to each address.

Yes, that's the best way to scan for devices, to do an empty write; ie just send the address with a write selected (lowest bit clear?) and then a stop condition.

finneyj, 2020-12-17:
Thanks @dpgeorge @martinwork 

You were quite right @martinwork. The code was stuck inside waitForStop(). The NRF52 hardware doesn't really like zero length write operations... it doesn't seem to have a mechanism to relay success - only error cases...

Anyhow, I've applied a patch here that introduces a special case workaround. This seems to work ok on the internal I2C bus on a v2 micro:bit using similar code to @martinwork above:
https://github.com/lancaster-university/codal-nrf52/commit/1a8624801d45301e0f5dfc2c1a17a5efca8082ab

Timeout is currently set to around 1ms, which seems about right (that would be 10 bytes at 100kHz bus speed). Happy to adjust if anyone has a better suggestion for the timeout period.

microbit-carlos, 2021-01-29:
Trying this in the current CODAL, with the external I2C bus without anything connected to it and I get all odd 7-bit addresses as "available":

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.serial.printf("Available addresses: ");
    // https://github.com/microbit-foundation/micropython-microbit-v2/blob/a76e1413bcd66f128a31d98756fc3d1f336d1580/src/codal_port/microbit_i2c.c#L69
    // 7-bit addresses 0b0000xxx and 0b1111xxx are reserved
    for (int addr = 0x08; addr < 0x78; ++addr) {
        int ret = uBit.i2c.write(addr << 1, NULL, 0, false);
        if (ret == MICROBIT_OK) {
            uBit.serial.printf("%d, ", addr);
        }
    }
    uBit.serial.printf("\n\n");
    for (;;);
}
```

```
Available addresses: 9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 31, 33, 35, 37, 39, 41, 43, 45, 47, 49, 51, 53, 55, 57, 59, 61, 63, 65, 67, 69, 71, 73, 75, 77, 79, 81, 83, 85, 87, 89, 91, 93, 95, 97, 99, 101, 103, 105, 107, 109, 111, 113, 115, 117, 119, 
```
Changing that to the internal `_i2c` instance and I get odd numbers until address 29, and all event numbers immediately after that.

```
Available addresses: 9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 30, 32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 60, 62, 64, 66, 68, 70, 72, 74, 76, 78, 80, 82, 84, 86, 88, 90, 92, 94, 96, 98, 100, 102, 104, 106, 108, 110, 112, 113, 114, 116, 118, 
```
This is with an LSM with addresses 0x19 (25) and 0x1E (30). So taking in consideration the count starts at 0x08 it looks like it's basically just failing at every second i2c write.

If I switch back to the external i2c, but change the starting point to 0x09, then we get all even numbers:
```
Available addresses: 10, 12, 14, 16, 18, 20, 22, 24, 26, 28, 30, 32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 60, 62, 64, 66, 68, 70, 72, 74, 76, 78, 80, 82, 84, 86, 88, 90, 92, 94, 96, 98, 100, 102, 104, 106, 108, 110, 112, 114, 116, 118, 
```

Running the I2c scan in MicroPython and we get a different result, where the "available" addresses change for each i2c.scan() call: 
- https://github.com/microbit-foundation/micropython-microbit-v2/issues/55#issuecomment-769941830
- https://github.com/microbit-foundation/micropython-microbit-v2/blob/a76e1413bcd66f128a31d98756fc3d1f336d1580/src/codal_port/microbit_i2c.c#L69

dpgeorge, 2021-02-08:
I can confirm the results that @microbit-carlos gets just above (using pur C++ code, a list of many addresses are available, both on `uBit.i2c` and `uBit._i2c` buses; using MicroPython, the available addresses change).

Adding a delay after the I2C write seems to make the problem go away.  The delay needs to be about 30us for the external bus, and 10us for the internal bus.  With such delays the available addresses are correct (this is for C++ as was as MicroPython code).  Any less of a delay and you start to see wrong addresses being reported as available, ie the I2C write succeeds when it should fail.

For completeness this is my test code:
```c++
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    for (int count = 0; count < 10; ++count) {
        uBit.serial.printf("Available addresses: ");
        // 7-bit addresses 0b0000xxx and 0b1111xxx are reserved
        for (int addr = 0x08; addr < 0x78; ++addr) {
            int ret = uBit.i2c.write(addr << 1, NULL, 0, false);
            system_timer_wait_us(30); // delay to workaround issue with above call
            if (ret == MICROBIT_OK) {
                uBit.serial.printf("%d, ", addr);
            }
        }
        uBit.serial.printf("\r\n");
    }   
    for (;;);
}
```

dpgeorge, 2021-02-08:
@finneyj is it part of the CODAL API that null (ie zero-length) byte writes are allowed on the I2C bus?  Well, it seems like it was on v1...

dpgeorge, 2021-02-08:
> Well, it seems like it was on v1...

Actually that's not entirely true.  MicroPython on v1 never used the DAL, instead we wrote a custom function to probe the bus.  See `i2c_probe()` in https://github.com/bbcmicrobit/micropython/commit/51be2575976fa6a0d9f509df357740da348903c2

martinwork, 2021-06-07:
Closing as all these samples, including @dpgeorge 's without a 30us delay, work now with i2c and _i2c.

closed 2021-06-07 by martinwork