lancaster-university/codal-microbit-v2 #50: Configuring pin 16 for events stops events on all pins

martinwork, 2020-12-09:
The sample below works fully on V1, can see events on all the pins and switch between edge and pulse.

On V2:

- If pin 16 is configured for events, none of the pins trigger events

- If the first mode chosen is for edge events, it can't switch to pulse

```
// This tests pin events on pins 0,1,2,8 and 12-16.
// Press A for pulse; B for edge.
// Pixels light in columns 3 and 4 when events are detected.
// The pull mode is UP, so connect pins to GND to trigger events.

#include "MicroBit.h"

#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif

MicroBit uBit;

void event_init_pin( int eventOn, int pinNumber);

void event_init_on( int eventOn)
{
    event_init_pin( eventOn, 0);
    event_init_pin( eventOn, 1);
    event_init_pin( eventOn, 2);
    event_init_pin( eventOn, 8);
    event_init_pin( eventOn, 12);
    event_init_pin( eventOn, 13);
    event_init_pin( eventOn, 14);
    event_init_pin( eventOn, 15);
    event_init_pin( eventOn, 16);
}

#if MICROBIT_CODAL
PullMode pullMode = PullMode::Up;
#else
PinMode  pullMode = PullUp;
#endif

bool eventHigh = false;
bool eventLow = false;

bool eventRise = false;
bool eventFall = false;

void onRise(MicroBitEvent e)
{
    eventRise = true;
}

void onFall(MicroBitEvent e)
{
    eventFall = true;
}

void onHi(MicroBitEvent e)
{
    eventHigh = true;
}

void onLo(MicroBitEvent e)
{
    eventLow = true;
}

void onButtonA(MicroBitEvent e)
{
    event_init_on( MICROBIT_PIN_EVENT_ON_PULSE);
}

void onButtonB(MicroBitEvent e)
{
    event_init_on( MICROBIT_PIN_EVENT_ON_EDGE);
}

void event_init_pin( int eventOn, int pinNumber)
{
    int pinID = MICROBIT_ID_IO_P0 + pinNumber;

    uBit.io.pin[ pinNumber].setPull( pullMode);

    uBit.io.pin[ pinNumber].eventOn( eventOn);

    uBit.messageBus.listen( pinID, MICROBIT_PIN_EVT_PULSE_HI, onHi);
    uBit.messageBus.listen( pinID, MICROBIT_PIN_EVT_PULSE_LO, onLo);

    uBit.messageBus.listen( pinID, MICROBIT_PIN_EVT_RISE, onRise);
    uBit.messageBus.listen( pinID, MICROBIT_PIN_EVT_FALL, onFall);
}

void event_init()
{
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK, onButtonB);
}

void event_fiber()
{
    while (true)
    {
        uBit.display.image.setPixelValue( 3, 0, eventHigh ? 255 : 0);
        uBit.display.image.setPixelValue( 3, 4, eventLow  ? 255 : 0);
        uBit.display.image.setPixelValue( 4, 0, eventRise ? 255 : 0);
        uBit.display.image.setPixelValue( 4, 4, eventFall ? 255 : 0);
        eventHigh = false;
        eventLow = false;
        eventRise = false;
        eventFall = false;
        uBit.sleep(200);
    }
}

void togglePixel( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue( x, y) ? 0 : 255);
}

void forever()
{
    while (true)
    {
        togglePixel( 0, 0);
        uBit.sleep(1000);
    }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    event_init();
    create_fiber( event_fiber);

    create_fiber( forever);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}
```

finneyj, 2020-12-09:
Thanks @martinwork 

Based on a very cursory glance before my next teaching, Edge connector P16 is the only pin in your test range that lives on the NRF52 GPIO PORT1 .... the others are on PORT0. So perhaps something in there is not mapping to PORTR correctly.

microbit-mark, 2021-01-21:
I'd like to prioritise this one if possible as it was reported by a makecode extension author.

tbeattyISB, 2021-01-28:
In case anyone is working on this, I am having the same issue on pin 2, except I'm using the microPython "Machine" library. I think it may be a problem with scheduling or interrupts. 

As an aside but likely related (I don't have time to explore), when the display is updated to clear then immediately light up the same LEDs in the array, the V2 bit flashes while the V1 bit stays steady.

microbit-mark, 2021-02-09:
Also reported by another person in https://support.microbit.org/a/tickets/43193

kboyd12345, 2021-02-17:
I raised the issue that Martin has mentioned above. I'm slightly confused about the meaning of 'edge' events and 'pulse' events here. When you say edge events, do you mean an interrupt? and if so, is this available in the makecode editor? (The person who reported this issue to me was originally looking for an interrupt and happened across the onPulsed() event)

martinwork, 2021-02-17:
@kboyd12345 See https://makecode.microbit.org/reference/pins/set-events

kboyd12345, 2021-02-17:
> @kboyd12345 See https://makecode.microbit.org/reference/pins/set-events

Martin, I realise this is not the place for this, but I wasn't aware that you could do this (although I was sure it must be possible). I've been looking at https://makecode.microbit.org/reference/pins and didn't see any way to register events for each pin. This is extremely useful - I'm surprised there isn't more written about it. I'd be interested to talk to you more about this.

martinwork, 2021-02-17:
@kboyd12345  I'm happy to help if I can. If it's about MakeCode, maybe https://github.com/microsoft/pxt-microbit/issues/3882 would be a place to start, for visibility with the MakeCode team?

martinwork, 2021-02-23:
@kboyd12345 For edge vs pulse distinction, see https://makecode.adafruit.com/learnsystem/pins-tutorial/digital-input/pin-events

finneyj, 2021-03-08:
I think this is fixed in this patch @martinwork :
https://github.com/lancaster-university/codal-nrf52/commit/dd77b5adab922b9dea92ffc882c30290c4139bd8

Can you please confirm?

finneyj, 2021-03-08:
p.s. Here's the test case I used, just for the record:

```
// This tests serial read and send

#include "MicroBit.h"

MicroBit uBit;

void onP0(Event e)
{
    if (e.value == DEVICE_PIN_EVT_RISE)
        DMESG("P0 HIGH: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_FALL)
        DMESG("P0 LOW: %d", (uint32_t)e.timestamp);
}

void onP16(Event e)
{
    if (e.value == DEVICE_PIN_EVT_RISE)
        DMESG("P16 HIGH: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_FALL)
        DMESG("P16 LOW: %d", (uint32_t)e.timestamp);
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.io.P0.setPull(PullMode::Down);
    uBit.io.P0.eventOn(DEVICE_PIN_EVENT_ON_EDGE);
    uBit.messageBus.listen(MICROBIT_ID_IO_P0, MICROBIT_EVT_ANY, onP0);

    uBit.io.P16.setPull(PullMode::Down);
    uBit.io.P16.eventOn(DEVICE_PIN_EVENT_ON_EDGE);
    uBit.messageBus.listen(MICROBIT_ID_IO_P16, MICROBIT_EVT_ANY, onP16);

    while(1)
    {
        uBit.display.image.setPixelValue(0,0,255);
        uBit.sleep(500);
        uBit.display.image.setPixelValue(0,0,0);
        uBit.sleep(500);
    }
}
```

martinwork, 2021-03-09:
Thanks @finney. Pin 16 is now working. 

But this problem remains... If the first event mode chosen after reset is for edge events, it can't switch to pulse. This corresponds to pressing B first in my sample. If A is pressed first (pulse), it can switch back and forth. Either order is OK with V1.

martinwork, 2021-03-09:
My fault! It's a separate issue really.

finneyj, 2021-03-09:
Fixed in https://github.com/lancaster-university/codal-nrf52/commit/7e94e76d78ff350f00edcd0d5ee34c4af5969a08

Would be good to run any edge/pulse tests that we have @martinwork, to ensure this nor works correctly in all cases we know about?

finneyj, 2021-03-09:
FYI, here's the test code I used:

```
#include "MicroBit.h"

MicroBit uBit;

void onP0(Event e)
{
    if (e.value == DEVICE_PIN_EVT_RISE)
        DMESG("P0 HIGH: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_FALL)
        DMESG("P0 LOW: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_PULSE_HI)
        DMESG("P0 HI_PULSE: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_PULSE_LO)
        DMESG("P0 LO_PULSE: %d", (uint32_t)e.timestamp);
}

void onP16(Event e)
{
    if (e.value == DEVICE_PIN_EVT_RISE)
        DMESG("P16 HIGH: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_FALL)
        DMESG("P16 LOW: %d", (uint32_t)e.timestamp);
        
    if (e.value == DEVICE_PIN_EVT_PULSE_HI)
        DMESG("P16 HI_PULSE: %d", (uint32_t)e.timestamp);

    if (e.value == DEVICE_PIN_EVT_PULSE_LO)
        DMESG("P16 LO_PULSE: %d", (uint32_t)e.timestamp);
}

void onButtA(Event)
{
    uBit.io.P0.eventOn(DEVICE_PIN_EVENT_ON_PULSE);
    uBit.io.P16.eventOn(DEVICE_PIN_EVENT_ON_PULSE);
}

void onButtB(Event)
{
    uBit.io.P0.eventOn(DEVICE_PIN_EVENT_ON_EDGE);
    uBit.io.P16.eventOn(DEVICE_PIN_EVENT_ON_EDGE);
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.io.P0.setPull(PullMode::Down);
    uBit.io.P0.eventOn(DEVICE_PIN_EVENT_ON_EDGE);
    uBit.messageBus.listen(MICROBIT_ID_IO_P0, MICROBIT_EVT_ANY, onP0);

    uBit.io.P16.setPull(PullMode::Down);
    uBit.io.P16.eventOn(DEVICE_PIN_EVENT_ON_EDGE);
    uBit.messageBus.listen(MICROBIT_ID_IO_P16, MICROBIT_EVT_ANY, onP16);

    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onButtA);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK, onButtB);

    while(1)
    {
        uBit.display.image.setPixelValue(0,0,255);
        uBit.sleep(500);
        uBit.display.image.setPixelValue(0,0,0);
        uBit.sleep(500);
    }
}
```

martinwork, 2021-03-14:
Thanks @finneyj, all working as expected.

My original sample above has a weird problem... If I now press A first, for pulse, it doesn't work, but I can't figure out why. I couldn't find a way to modify your test to make it go wrong. Almost any modification to my sample makes it start working. Adding "volatile" to the global eventXXX variables doesn't help, but moving the initialisation of the onButton click handlers does.

closed 2021-03-14 by martinwork

finneyj, 2021-03-14:
Thanks @martinwork. That corner case is odd... don't know why, sorry.

finneyj, 2021-03-16:
Fixed in v0.2.25