lancaster-university/codal-microbit-v2 #53: Buttons A and B fire HOLD events only once per button. AB events do not always show if A or B listener is registered

martinwork, 2020-12-12:
The code below seems to work as expected on V1, though I'm not sure exactly how it should work. 

V2 sees HOLD events only once per button.

If an AB listener is also registered (commented out below), V2 also sees LONG_CLICK events only once per button, and doesn't see any AB events.

If only an AB listener is registered, V2 sees AB events but HOLD and LONG_CLICK fire only once.

If AB is registered with one of A or B, AB events depend on the order A and B are pressed and released. This is the case on V1 and V2, but the exact behaviours are different: AB up or down missing; or AB click missing.

If all three listeners are registered V1 sees HOLD events for A and B, but not AB.

```
// This tests button events
// See serial output in a terminal program 

#include "MicroBit.h"

MicroBit uBit;

void onButtonEvent(MicroBitEvent e)
{
  ManagedString btnName;
  ManagedString btnEvent;

  switch ( e.source)
  {
    case MICROBIT_ID_BUTTON_A:  btnName = "A "; break;
    case MICROBIT_ID_BUTTON_B:  btnName = " B"; break;
    case MICROBIT_ID_BUTTON_AB: btnName = "AB"; break;
  }

  switch ( e.value)
  {
    case MICROBIT_BUTTON_EVT_CLICK:         btnEvent = "CLICK";         break;
    case MICROBIT_BUTTON_EVT_DOUBLE_CLICK:  btnEvent = "DOUBLE_CLICK";  break;
    case MICROBIT_BUTTON_EVT_DOWN:          btnEvent = "DOWN";          break;
    case MICROBIT_BUTTON_EVT_HOLD:          btnEvent = "HOLD";          break;
    case MICROBIT_BUTTON_EVT_LONG_CLICK:    btnEvent = "LONG_CLICK";    break;
    case MICROBIT_BUTTON_EVT_UP:            btnEvent = "UP";            break;
  }

  uBit.serial.send( btnName + " " + btnEvent + "\n");
}

void onButtonA(MicroBitEvent e)
{
  ManagedString btnEvent;
  ManagedString btnName = "A ";

  switch ( e.value)
  {
    case MICROBIT_BUTTON_EVT_CLICK:         btnEvent = "CLICK";         break;
    case MICROBIT_BUTTON_EVT_DOUBLE_CLICK:  btnEvent = "DOUBLE_CLICK";  break;
    case MICROBIT_BUTTON_EVT_DOWN:          btnEvent = "DOWN";          break;
    case MICROBIT_BUTTON_EVT_HOLD:          btnEvent = "HOLD";          break;
    case MICROBIT_BUTTON_EVT_LONG_CLICK:    btnEvent = "LONG_CLICK";    break;
    case MICROBIT_BUTTON_EVT_UP:            btnEvent = "UP";            break;
  }

  uBit.serial.send( btnName + " " + btnEvent + "\n");
}

void onButtonB(MicroBitEvent e)
{
  ManagedString btnEvent;
  ManagedString btnName = " B";

  switch ( e.value)
  {
    case MICROBIT_BUTTON_EVT_CLICK:         btnEvent = "CLICK";         break;
    case MICROBIT_BUTTON_EVT_DOUBLE_CLICK:  btnEvent = "DOUBLE_CLICK";  break;
    case MICROBIT_BUTTON_EVT_DOWN:          btnEvent = "DOWN";          break;
    case MICROBIT_BUTTON_EVT_HOLD:          btnEvent = "HOLD";          break;
    case MICROBIT_BUTTON_EVT_LONG_CLICK:    btnEvent = "LONG_CLICK";    break;
    case MICROBIT_BUTTON_EVT_UP:            btnEvent = "UP";            break;
  }

  uBit.serial.send( btnName + " " + btnEvent + "\n");
}

void onButtonAB(MicroBitEvent e)
{
  ManagedString btnEvent;
  ManagedString btnName = "AB";

  switch ( e.value)
  {
    case MICROBIT_BUTTON_EVT_CLICK:         btnEvent = "CLICK";         break;
    case MICROBIT_BUTTON_EVT_DOUBLE_CLICK:  btnEvent = "DOUBLE_CLICK";  break;
    case MICROBIT_BUTTON_EVT_DOWN:          btnEvent = "DOWN";          break;
    case MICROBIT_BUTTON_EVT_HOLD:          btnEvent = "HOLD";          break;
    case MICROBIT_BUTTON_EVT_LONG_CLICK:    btnEvent = "LONG_CLICK";    break;
    case MICROBIT_BUTTON_EVT_UP:            btnEvent = "UP";            break;
  }

  uBit.serial.send( btnName + " " + btnEvent + "\n");
}

void togglePixel( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue( x, y) ? 0 : 255);
}

void forever()
{
    while (true)
    {
        togglePixel( 0, 0);
        uBit.sleep(1000);
    }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.messageBus.listen( MICROBIT_ID_BUTTON_A,  MICROBIT_EVT_ANY, onButtonA);
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_B,  MICROBIT_EVT_ANY, onButtonB);
    //uBit.messageBus.listen( MICROBIT_ID_BUTTON_AB, MICROBIT_EVT_ANY, onButtonAB);

    create_fiber( forever);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}
```

finneyj, 2021-03-13:
Thanks @martinwork. 

Fixed in https://github.com/lancaster-university/codal-core/commit/37d6f5e4c42fa6e65a2763bd89df22bb6a9a8c62

martinwork, 2021-03-14:
Thanks @finneyj Fixed with one line!

closed 2021-03-14 by martinwork

finneyj, 2021-03-14:
Thanks @martinwork . Yep - the best ones are always fixed in one line!

finneyj, 2021-03-16:
Fixed in v0.2.25