lancaster-university/codal-microbit-v2 #55: Garbage characters in UART RX buffer on power up

dpgeorge, 2020-12-14:
After a power up (but not reset) the UART RX buffer seems to be full of garbage (or at least unwanted) characters.  And the number of garbage characters seems to be exactly the size of the RX buffer (eg 20 chars for the default size).

To reproduce:
```c
int main() {
    uBit.init();
  
    const char *hexchr = "0123456789abcdef";
    for (;;) {
        int c = uBit.serial.read(SYNC_SPINWAIT);
        char str[60];
        memcpy(str, "RX(--)\r\n", 8);
        str[3] = hexchr[(c >> 4) & 0xf];
        str[4] = hexchr[c & 0xf];
        uBit.serial.send((uint8_t *)str, 8, SYNC_SPINWAIT);
    }

    return 0;
}
```
Running that on power up will usually show the issue (the microbit needs to be fully discharged before plugging in to USB to see the problem).

Attaching to the USB serial port after power up and pressing space many times I get the following output:
```
RX(03) <- this is printed when "space" is pressed the first time
RX(fd) <- this is printed when "space" is pressed the second time
RX(03)
RX(f5)
RX(21)
RX(02)
RX(00)
RX(47)
RX(89)
RX(e0)
RX(00)
RX(00)
RX(00)
RX(00)
RX(00)
RX(84)
RX(5e)
RX(00)
RX(20)
RX(db) <- this is the last garbage character in the 20-byte buffer
RX(20) <- this is printed when "space" is pressed the 21st time
RX(20)
RX(20)
RX(20)
RX(20)
RX(20)
...
```
I suspect this is the same as #45 but raised a separate issue because it's not just one garbage character but many (the size of the RX buffer), and doesn't need `setRxBufferSize` to trigger it.

finneyj, 2020-12-15:
Many thanks for the detailed repro case @dpgeorge @martinwork @microbit-carlos 

I've just looked into this, and I think I know what's going on here. After a cold start on a v2 micro:bit, the UART hardware is reporting a phenomenal number of framing errors / break conditions. I suspect the KL27 interface chip is holding the line low as it initializes from a cold start... which is why we don't see this behaviour on a warm reset. 

You can test this by introducing a long delay (I used 5 seconds) before entering damien's test case above - in which case the test performs as expected.

I've update the error handling code in the NRF52Serial driver so that it is a bit more tolerant to these conditions, which seems to clear up this issue. Once i've had chance to do a little testing/code review alongside the other serial issues that have been raised I'll apply that patch to master.

microbit-carlos, 2021-03-08:
We don't see this specific uart issue anymore in MicroPython, we can close this ticket. Thanks Joe!

closed 2021-03-08 by microbit-carlos