lancaster-university/codal-microbit-v2 #58: Touch pin triggered when mic casing is touched

microbit-mark, 2020-12-17:
Using a simple touch program in MakeCode:

https://makecode.microbit.org/_a5aHFuA9eXfY

1. Download program
2. Touch logo
3. Reset
4. Touch mic casing on back of board. Logo touch is triggered

We raised this initially in 
https://github.com/microbit-foundation/codal-microbit/issues/130 and the calibration was fixed for touching around the nRF52

finneyj, 2020-12-17:
Is the reset step important @microbit-mark ?

finneyj, 2020-12-17:
I can't repro on my v with this, but I can trigger it is I wiggle enough around the NRF52 just next to the microphone casing.

```
#include "MicroBit.h"

MicroBit uBit;

void onTouchFace2(MicroBitEvent e)
{
    uBit.display.print("J");
}

int main()
{
    uBit.init();
    uBit.messageBus.listen(MICROBIT_ID_FACE, MICROBIT_BUTTON_EVT_CLICK, onTouchFace2);

    release_fiber();
}
```

finneyj, 2020-12-17:
We can of course adjust the default sensitivity, but I think we need a process to decide what that should be... or we'll just keep chasing our tails around... It is also important to test on both battery and USB power - the resistive component associated with a well grounded USB power supply will increase sensitivity substantially,

There is already an API to adjust the sensitivity:
https://github.com/lancaster-university/codal-core/blob/master/source/drivers/TouchButton.cpp#L91

The current default is set to 3500 (higher values are less sensitive) is defined here: 
https://github.com/lancaster-university/codal-microbit-v2/blob/master/target.json#L42

microbit-mark, 2021-01-11:
> Is the reset step important @microbit-mark ?

No. This is just to reset the MakeCode program.

> There is already an API to adjust the sensitivity:
https://github.com/lancaster-university/codal-core/blob/master/source/drivers/TouchButton.cpp#L91

Perhaps if there is an API we should look to expose this as an advanced feature in the editors. @pelikhan @abchatra for ref.

microbit-mark, 2021-04-26:
> micro:bit support ticket #44922

This has also been noted in the wild by a user asking if it is a bug. I've tested this again on USB and battery and updated the ticket to note that it appears to be the connection around the casing not the mic casing itself that cause the issue, so I expect this is the same as https://github.com/microbit-foundation/codal-microbit/issues/130

microbit-mark, 2022-05-06:
@microbit-carlos Not often. I had another issue recently where the logo appeared to be triggered on start, without touching it so waiting to see if it is similar. Perhaps we should try it with a bunch of people to see if it's easily triggered.

microbit-carlos, 2024-05-07:
We haven't really got more reports about this since the calibration/sensitivity was changed, so we can close this as completed.

closed 2024-05-07 by microbit-carlos

jaustin, 2024-05-07:
Reminded me to file https://github.com/microsoft/pxt-microbit/issues/5653