lancaster-university/codal-microbit-v2 #59: When the UART RX pin is grounded it can halt the user programme

microbit-carlos, 2020-12-18:
By routing the UART pins to the edge connector, and then grounding the RX pin the display stops updating.

It doesn't look like it crashes, as usually that would leave a column/row light up.

In this example the text is scrolling and by tapping pin1 with a jumper cable to ground you can easily (not on the first tap, but you can try a few times) clear the display and nothing else seems to happen:

```
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();

    uBit.serial.redirect(uBit.io.P0, uBit.io.P1);
    uBit.serial.setBaud(115200);

    while (1) {
        uBit.display.scroll("HELLO WORLD!", 400);
    }
}
```

microbit-mark, 2021-01-11:
> micro:bit support #42575

This problem is affecting accessory users and has been raised by @lionyhw on behalf of Elecfreaks.

"The problem that serial port can't be used after reset needs to be solved."

@finneyj Can we prioritise a fix?

finneyj, 2021-03-04:
Hi @microbit-carlos  @microbit-mark. 

I just tried to reproduce this, but it seems to behave ok against the head revision of codal-microbit-v2 and codal-nrf52. I don't seem to be able to reproduce the behaviour described above. Can either of you verify that this is still an issue? 

Perhaps it may have been already fixed in this commit?
https://github.com/lancaster-university/codal-nrf52/commit/efd63319af7c5f8034a4be120cf9bd781ba7c3d2

microbit-carlos, 2021-03-08:
Yeah, I cannot replicate this anymore in CODAL not current MakeCode, so I think we can close this.

@microbit-mark can you ask elecfreaks to test again? It should work now, but if it still doesn't we can reopen this issue.

closed 2021-03-08 by microbit-carlos

finneyj, 2021-03-16:
Fixed in v0.2.25