lancaster-university/codal-microbit-v2 #60: serial.redirect breaks serial.

microbit-carlos, 2020-12-21:
From https://github.com/microbit-foundation/micropython-microbit-v2/issues/29#issuecomment-748710061:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();

    for (int i = 0; i < 10; ++i) {
        char x = uBit.serial.read();
        uBit.serial.sendChar(x);
    }

    uBit.serial.setBaud(115200);
    uBit.serial.redirect(uBit.io.usbTx, uBit.io.usbRx);

    while (1) {
        char x = uBit.serial.read();
        uBit.serial.sendChar(x);
    }
}
```
In the above test, the first 10 chars are echoed back correctly, but then when it redirects (to the same pins) following chars are not.

microbit-mark, 2021-01-11:
> micro:bit support #42575

This problem is affecting accessory users and has been raised by @lionyhw on behalf of elecfreaks.

"The problem that serial port can't be used after reset needs to be solved."

Can we prioritise a fix?

jaustin, 2021-01-20:
@finneyj this is from a live issue in MicroPython

martinwork, 2021-02-10:
The micro:bit appears to stop responding after about 35 characters - see expanded test below. After the redirect, serial.send is still working, but serial.read returns wrong characters, mostly zero, apparently the same characters whatever is sent.

Oddly, if I edit Serial::redirect to remove the disableInterrupt & enableInterrupt calls, it's continues to receive and echo the right characters.

```
#include "MicroBit.h"

MicroBit uBit;

void togglePixel( int x, int y);

void test()
{
    for (int i = 0; i < 10; ++i) {
        char x = uBit.serial.read();
        uBit.display.printChar(x);
        uBit.serial.sendChar(x);
    }

    uBit.serial.setBaud(115200);
    uBit.serial.redirect(uBit.io.usbTx, uBit.io.usbRx);

    while (1) {
        togglePixel( 4, 0);
        char x = uBit.serial.read();

        ManagedString s((int)x);

        if ( x > 31 && x < 126)
          uBit.display.printChar(x);
        else
          uBit.display.scroll(s);

        togglePixel( 4, 1);

        if ( x > 31 && x < 126)
          uBit.serial.sendChar(x);
        else
        {
          s = "CODE=" + s;
          uBit.serial.send(s);
        }

        togglePixel( 4, 2);
    }
}

void togglePixel( int x, int y)
{
    uBit.display.image.setPixelValue( x, y, uBit.display.image.getPixelValue( x, y) ? 0 : 255);
}

void forever()
{
    while (true)
    {
        togglePixel( 0, 0);
        uBit.sleep(1000);
    }
}

int  main() {
    uBit.init();

    create_fiber( forever);
    create_fiber( test);

    release_fiber();
}

```

martinwork, 2021-02-10:
I think the value of bytesProcessed is wrong after enableInterrupt().

It stops it crashing if I insert `bytesProcessed = 0;`
in enableInterrupt(), before this line...
https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Serial.cpp#L155

I'm pretty sure of the problem, but I'm not sure this is the solution!

Maybe disableInterrupt() should call updateRxBufferAfterENDRX()?

@finneyj ?

martinwork, 2021-02-10:
initialiseRx() calls enableInterrupt, so maybe this test should be just "else" or the following block would be executed twice.
https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Serial.cpp#L154

jaustin, 2021-02-12:
@finneyj we have a MicroPython release ticket blocked on this issue so could you please look at this one as a priority?

martinwork, 2021-02-12:
The test programs make more sense if dmaBuffer is filled with "ABCDEFGHIJKLMNOPQRSTUVWXYZabcdef" just before:
https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Serial.cpp#L155
Then just type digits. After redirect the echo is KLMNOPQRSTUVWXYZabcdef plus a few odd characters before crash.

finneyj, 2021-02-12:
ok - thanks. I'll try to look at it this weekend.

finneyj, 2021-03-05:
Thanks for the analysis @martinwork. I agree that we need to reset `bytesProcessed` when pins are redirected due to the fact the code stops and restarts the UARTE RX hardware here; https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Serial.cpp#L158

When that happens, the NRF52 hardware will start filling the buffer again from the beginning. I agree  the best fix is to apply the reset of `bytesProcessed` in that code block somewhere just before this line, so that we're free of race conditions from the IRQ.

I've also got a working fix based on this approach. I'll work this up into the master branch.

finneyj, 2021-03-08:
Fixed in https://github.com/lancaster-university/codal-nrf52/commit/4d081080aa402b6399c070ee8882c7e7a7f57fa4

closed 2021-03-08 by finneyj

microbit-carlos, 2021-03-08:
Thanks Joe! Can confirm this fixes it for me as well.

finneyj, 2021-03-16:
Fixed in v0.2.25