lancaster-university/codal-microbit-v2 #61: Build error on Ubuntu 20.04

r8d8, 2020-12-28:
Trying to build project on Ubuntu 20.04, using CMAKE 3.17 and having next error:
```bash
[main] Building folder: codal-microbit-v2 
[main] Configuring folder: codal-microbit-v2 
[proc] Executing command: /usr/local/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -H/home/k2/Workbench/codal-microbit-v2 -B/home/k2/Workbench/codal-microbit-v2/build -G "Unix Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Configuring incomplete, errors occurred!
[cmake] See also "/home/k2/Workbench/codal-microbit-v2/build/CMakeFiles/CMakeOutput.log".
[cmake] CMake Error at CMakeLists.txt:5 (RECURSIVE_FIND_DIR):
[cmake]   Unknown CMake command "RECURSIVE_FIND_DIR".
```

microbit-carlos, 2021-01-04:
Hi @r8d8,

Could you provide the full terminal output from the moment the build command is executed? It provides a lot of info that will be useful for debugging.

microbit-carlos, 2021-01-20:
Hi @r8d8, have you been able to resolve this issue?

r8d8, 2021-01-20:
Hi @microbit-carlos 

Have moved to the `samples-v2` repo and build.py script in samples repo has succeeded
Scripts rely on CODAL runtime, so I suppose initial build error was my local env.
If I would figure out solution to these issue, will comment later.

Thanks for your attention

closed 2021-01-20 by r8d8