lancaster-university/codal-microbit-v2 #63: Linking fails with GCC 10 building Edge Impulse sample

jaustin, 2021-01-04:
I'm proxy-filing this as the Edge Impulse team pointed it out to us.

https://forum.edgeimpulse.com/t/voice-activated-micro-bit-with-machine-learning/811

Looks like the linker doesn't like "--output-def" and they're using GCC 10.2.1. Same user can build with GNU Arm Embedded Toolchain 9-2020-q2-update

Has anyone else encountered this? @finneyj or @jamesadevine is either of you using latest toolchains?

martinwork, 2021-04-20:
I've installed
`gcc-arm-none-eabi-10-2020-q4-major-win32.exe`
and built
`https://github.com/edgeimpulse/voice-activated-microbit`
with
`python build.py`

It's odd that the output https://forum.edgeimpulse.com/t/voice-activated-micro-bit-with-machine-learning/811/2 looks like it's trying to build a DLL, which is related to "--output-def".

The text "output-def" is not found anywhere in my `voice-activated-microbit` folder

```
C:\github\voice-activated-microbit>arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

C:\github\voice-activated-microbit>python build.py
codal-microbit-v2 is already installed
Set target: codal-microbit-v2
Using target.json (dev version)
Targeting codal-microbit-v2
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
Supressing -Wexpansion-to-defined.
-- Adding library path: (C:/github/voice-activated-microbit/libraries)
Installing dependencies...
codal-core is already installed
codal-nrf52 is already installed
codal-microbit-nrf5sdk is already installed
Using library: codal-microbit-v2
Using library: codal-core
Using library: codal-nrf52
Using library: codal-microbit-nrf5sdk
-- Configuring done
-- Generating done
-- Build files have been written to: C:/github/voice-activated-microbit/build
[90/711] Building CXX object CMakeFiles/MICROBIT.dir/source/ContinuousAudioStreamer.cpp.obj

[etc.]

[687/711] Building C object libraries/codal-microbit-nrf5sdk/CMakeFiles/codal-microbit-nrf5sdk.dir/nRF5SDK/components/ble/peer_manager/gatt_cache_manager.c.obj
../libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/peer_manager/gatt_cache_manager.c: In function 'service_changed_send_in_evt':
../libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/peer_manager/gatt_cache_manager.c:364:9: warning: this statement may fall through [-Wimplicit-fallthrough=]
  364 |         {
      |         ^
../libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/peer_manager/gatt_cache_manager.c:385:9: note: here
  385 |         case NRF_ERROR_NOT_SUPPORTED:
      |         ^~~~
[711/711] converting to bin file.

C:\github\voice-activated-microbit>

```

microbit-carlos, 2021-04-21:
@martinwork just to confirm, it builds successfully in your set up with GCC 10.2.1, right?

martinwork, 2021-04-21:
@microbit-carlos Yes. I didn't change anything except installing gcc-arm-none-eabi-10-2020-q4-major-win32.exe, so the compiler identified as 10.2.1. I have gone back to 9.3.1 for now!

martinwork, 2021-04-24:
Maybe something to do with the cmake version (mine is 3.18.4)?

Is variable BUILD_SHARED_LIBS being set ON?

microbit-carlos, 2023-01-17:
We are unable to replicate and it looks like it might be an issue with the dev environment from the original report in that forum. As that thread is now more of "catch all" for several build issues that are not directly related to CODAL, I think we can close this one until we have a specific problem to debug.

closed 2023-01-17 by microbit-carlos