lancaster-university/codal-microbit-v2 #66: DAL PinMode not available

microbit-carlos, 2021-01-18:
This will compile with DAL and fail with CODAL:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();
    uBit.io.P0.setPull(PullUp);
}
```

```
/Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/source/main.cpp: In function 'int main()':
/Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/source/main.cpp:8:30: error: no matching function for call to 'codal::NRF52Pin::setPull(PinMode)'
     uBit.io.P0.setPull(PullUp);
                              ^
In file included from /Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-microbit-v2/model/MicroBitIO.h:29:0,
                 from /Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-microbit-v2/inc/compat/MicroBitCompat.h:45,
                 from /Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-microbit-v2/inc/MicroBitConfig.h:6,
                 from /Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-microbit-v2/model/MicroBit.h:28,
                 from /Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/source/main.cpp:1:
/Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-nrf52/inc/NRF52Pin.h:358:21: note: candidate: virtual int codal::NRF52Pin::setPull(codal::PullMode)
         virtual int setPull(PullMode pull) override;
                     ^~~~~~~
/Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-nrf52/inc/NRF52Pin.h:358:21: note:   no known conversion for argument 1 from 'PinMode' to 'codal::PullMode'
```

Need to use `PullMode::Up` instead, but it's not DAL compatible.

martinwork, 2021-04-21:
To use DAL PinMode and PullXXX in CODAL, could we replace in MicroBitCompat.h
```
typedef enum {
    PullNone = 0,
    PullDown = 1,
    PullUp = 3,
    PullDefault = PullUp
} PinMode;
```
with
```
typedef codal::PullMode PinMode;
const codal::PullMode PullNone    = codal::PullMode::None;
const codal::PullMode PullDown    = codal::PullMode::Down;
const codal::PullMode PullUp      = codal::PullMode::Up;
const codal::PullMode PullDefault = codal::PullMode::Up;
```
With this the MicroBitButton constructors don't need getPullModeFronPinMode.

Is there a problem with the fact that CODAL PullMode::Up==2 while DAL PullUp==3?

martinwork, 2021-04-22:
This sample wraps a PinMode in a PullMode class, so that V2 CODAL style PullMode code compiles for V1 DAL.

It should start in PullUp mode, then B sets PullMode::Down, A sets PullMode::Up.

```
#include "MicroBit.h"

#ifndef MICROBIT_CODAL
#ifdef CODAL_CONFIG_H
#define MICROBIT_CODAL 1
#else
#define MICROBIT_CODAL 0
#endif
#endif

#if MICROBIT_CODAL

#else // MICROBIT_CODAL

class PullMode
{
public:
    PinMode pm;

    PullMode()            : pm( PullNone) {};
    PullMode( uint8_t p)  : pm( (PinMode)p) {};
    PullMode( PinMode p)  : pm( p) {};

    static const PinMode None = PullNone;
    static const PinMode Down = PullDown;
    static const PinMode Up   = PullUp;

    operator PinMode() { return pm; }
    operator uint8_t() { return pm; }
};

#endif // MICROBIT_CODAL

MicroBit uBit;

void onButtonA(MicroBitEvent e)
{
    PullMode pullmode = PullMode::Up;
    uBit.io.P0.setPull(pullmode);
}

void onButtonB(MicroBitEvent e)
{
    uBit.io.P0.setPull(PullMode::Down);
}

void display()
{
   while (true)
   {
     uBit.display.printChar( uBit.io.P0.getDigitalValue() ? '1' : '0');
     uBit.sleep(100);
   }
}

void DoPullMode( PullMode p)
{
    uBit.io.P0.setPull(p);
}

void DoPinMode( PinMode p)
{
    uBit.io.P0.setPull(p);
}

int  main() {
    uBit.init();

    uBit.messageBus.listen( MICROBIT_ID_BUTTON_A,  MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_B,  MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    uBit.io.P0.setPull(PullUp);
    uBit.io.P0.setPull(PullMode::Up);

    DoPinMode( PullUp);
    DoPullMode( PullMode::Up);

    PullMode pullmode = PullUp;

    DoPinMode( pullmode);
    DoPullMode( pullmode);

    PullMode pinmode = PullMode::Up;

    DoPinMode( pinmode);
    DoPullMode( pinmode);

    ManagedString s1( (int) sizeof(pullmode));
    uBit.display.scroll( s1);

    ManagedString s2( (int) sizeof(pinmode));
    uBit.display.scroll( s2);

    create_fiber( display);

    release_fiber();
}
```