lancaster-university/codal-microbit-v2 #67: Floating pin reading as high (DAL has pull downs)

microbit-carlos, 2021-01-19:
This programme in V1 DAL always reads the pin a floating pin as "low", until a jumper cable is used to connect it to the 3V pin.

```cpp
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();
    // uBit.io.P1.setPull(PullMode::Down);
    while (1) {
        if (uBit.io.P1.getDigitalValue() == 1) {
            uBit.display.printChar('H');
            uBit.sleep(400);
        } else {
            uBit.display.printChar('L');
            uBit.sleep(400);
        }
    }
}
```

The sleep is not required, it also shows that in V2 CODAL the first reading of the pin is always `0`, as on reset it always it shows the `L` on the display once and then stays `H`.

Uncommenting the pull down matches the V1 DAL behaviour.

martinwork, 2021-02-22:
Related: https://github.com/lancaster-university/codal-microbit-v2/issues/17

microbit-carlos, 2021-02-22:
Thanks Martin, didn't see that issue when I opened this. As it is essentially the same I'll close this one to concentrate the conversation over there.

closed 2021-02-22 by microbit-carlos