lancaster-university/codal-microbit-v2 #68: evt.timestamp Zero on Codal

I-Programmer-info, 2021-01-20:
Same program run on V1 gives a reasonable event timestamp. On V2 it returns zero.

#include "MicroBit.h"
MicroBit uBit;
uint64_t timenow = 1;

void onPulseEdge(MicroBitEvent evt)
{
    timenow = evt.timestamp;
    uBit.io.P0.eventOn(MICROBIT_PIN_EVENT_NONE);
}

int main()
{
    uBit.init();
    uBit.io.P0.eventOn(MICROBIT_PIN_EVENT_ON_EDGE);
    uBit.messageBus.listen(MICROBIT_ID_IO_P0, MICROBIT_PIN_EVT_RISE, onPulseEdge,
                           MESSAGE_BUS_LISTENER_IMMEDIATE);
    uBit.serial.printf("Ready \n\r");
    uBit.sleep(5000);
    uBit.serial.printf("%d\n\r", (int)timenow);
}

I-Programmer-info, 2021-01-31:
I have found the "solution" to this particular problem but I don't know if it occurs in any other device event code. Can someone who knows the code base better than me (almost anyone!) please fix?
In libraries\codal-nrf52\source\NRF52Pin.cpp change two methods:
void NRF52Pin::rise()
{
    if(status & IO_STATUS_EVENT_PULSE_ON_EDGE)
        pulseWidthEvent(DEVICE_PIN_EVT_PULSE_LO);
    if(status & IO_STATUS_EVENT_ON_EDGE)
**//Change     Event(id, DEVICE_PIN_EVT_RISE, 0); to**
        Event(id, DEVICE_PIN_EVT_RISE, system_timer_current_time_us());
    if (status & IO_STATUS_INTERRUPT_ON_EDGE && gpio_irq)
        this->gpio_irq(1);
}
void NRF52Pin::fall()
{
    if(status & IO_STATUS_EVENT_PULSE_ON_EDGE)
        pulseWidthEvent(DEVICE_PIN_EVT_PULSE_HI);
    if(status & IO_STATUS_EVENT_ON_EDGE)
**// Change  Event(id, DEVICE_PIN_EVT_FALL,0 ); to**
        Event(id, DEVICE_PIN_EVT_FALL, system_timer_current_time_us());
    if (status & IO_STATUS_INTERRUPT_ON_EDGE && gpio_irq)
        this->gpio_irq(0);
}

finneyj, 2021-02-12:
Hi @I-Programmer-info,

Thanks so much for raising this and providing the suggested fix. Apologies we couldn't get back to you sooner. Universities are really swamped at the moment, so not much time for voluntary work I'm afraid! :(

I think this is indeed pretty much the correct fix. It's actually even easier - the timestamp is an optional parameter, so if omitted, the event will automatically insert the current time using system_timer_current_time_us(). https://github.com/lancaster-university/codal-core/blob/master/source/types/Event.cpp#L66

I went back through the git commit logs, and can't find any explicit reason why NRF52Pin::rise / NRF52Pin::fall were set up like this. My only thought is that someone was trying to reduce interrupt latency/response time so reducing code complexity. We should maintain compatibility with microbit-dal v1 though. So at least for micro:bit v2, we should patch this. 

I think I'll make it a compile time option though (enabled by default for micro:bit), so that we don't affect the behaviour of any other nrf52 based devices. How does that sound?

finneyj, 2021-03-08:
Fixed in https://github.com/lancaster-university/codal-nrf52/commit/35efdba612361c9a26e13e16ba64c9beb04e65e0

OK @I-Programmer-info ?

finneyj, 2021-03-16:
Fixed in v0.2.25

closed 2022-05-05 by microbit-carlos