lancaster-university/codal-microbit-v2 #69: how can i use spi

hnwangkg-ezio, 2021-01-22:


hnwangkg-ezio, 2021-01-22:
#include "mbed.h"

SPI lcd_spi(MOSI, MISO, SCK);
lcd_spi.format(8,0);
lcd_spi.frequency(9000000);

right?

I-Programmer-info, 2021-01-27:
A loop back example for V2
#include "MicroBit.h"
MicroBit uBit;

SPI spi(MOSI, MISO, SCK);

MicroBitPin CS(MICROBIT_ID_IO_P16, MICROBIT_PIN_P16, PIN_CAPABILITY_DIGITAL);
int main()
{
    CS.setDigitalValue(1);
    spi.format(8, 0);
    spi.frequency(1000000);
    uint8_t buf[1] = {0xAA};
    while (1)
    {
        CS.setDigitalValue(0);
        spi.transfer(buf, 1, buf, 1); 
        CS.setDigitalValue(1);
        uBit.serial.printf("data %X\n\r", buf[0]);
        if (buf[0] == 0xAA)
            uBit.serial.printf("data received correctly\n\r");
    };
    release_fiber();
    return 0;
}

finneyj, 2021-02-12:
Hi @hnwangkg-ezio 

Did this help you to solve your problem?