lancaster-university/codal-microbit-v2 #76: How to use standard Nordic tools to flash a micro:bit V2 via Bluetooth

martinwork, 2021-02-01:
It might be useful to document ways to flash a hex to V2 via Bluetooth using standard tools.

jaustin, 2022-05-05:
Hi @martinwork - I think you're right that this would be useful - do you already have instructions, perhaps in an email or something? If so could you share them here so @JohnVidler can work them into the CODAL documentation?

martinwork, 2022-05-05:
@jaustin I don't think I have anything, I'm afraid. I think the first step would be to split the hex into softdevice, app and bootloader, that can then be packaged for nRF Connect. These package scripts will have clues: https://github.com/microbit-foundation/v2-bootloader/tree/master/scripts. Do we have a tool to split a universal hex? A utility that builds an nRF Connect zip from a universal hex might be more helpful than step by step instructions.