lancaster-university/codal-microbit-v2 #78: BLE device name changes after connection/pairing

ukBaz, 2021-02-04:
I am using a micro:bit V2 with MakeCode blocks. I am mainly connecting to it from a Raspberry Pi and I've noticed that at some point it loses the unique name in the brackets and just changes to being called micro:bit. For example, here is the v2 during scanning with `bluetoothctl` on the Raspberry Pi
```
Device E9:06:4D:45:FC:8D BBC micro:bit [tetog]
```
After pairing it is still named:
```
Device E9:06:4D:45:FC:8D BBC micro:bit [tetog]
```
Then after I connect it gets changed:
```
[bluetooth]# connect E9:06:4D:45:FC:8D 
Attempting to connect to E9:06:4D:45:FC:8D
[CHG] Device E9:06:4D:45:FC:8D Connected: yes
[CHG] Device E9:06:4D:45:FC:8D Name: BBC micro:bit
[CHG] Device E9:06:4D:45:FC:8D Alias: BBC micro:bit
```
And when I do future listings I don't see the unique name anymore:
```
Device E9:06:4D:45:FC:8D BBC micro:bit
```
I have done a test with `nRF Connect` on Android phone and I can see:
 - `0x1800` Primary Service
 - `0x2A00` Device Name = `BBC micro:bit` (Properties: READ)

If I do the same in `nRF Connect` with a V1 I see:
 - `0x1800` Primary Service
 - `0x2A00` Device Name = `BBC micro:bit [zuvig] (Properties: READ, WRITE)

It would be good if the name could be persistent on the v2 as it is on the v1.

@martinwork

jaustin, 2021-02-08:
@martinwork do you mind taking a quick look at this please?

martinwork, 2021-02-08:
I don't think the full name would normally be persistent for either V1 or V2. I would expect the Device Name to change depending on whether the last connection was in pairing mode, normal mode or DFU mode. The full name is usually advertised only in pairing mode, for security reasons.

When I look on Android, V1 and V2 seem to behave the same. Having uses nRF Connect to pair with and then re-connect to V1 and V2 in non-pairing mode, the Device Name characteristic value that nRF Connect reads, and the names shown in Settings/Connections/Bluetooth/PAIRED DEVICES, are the short name. Note that nRF Connect caches the name in its tabs and is lazy about updating it.

In DAL and CODAL, the [xxxxx] is added to the advertising name in pairing mode, and when MICROBIT_BLE_WHITELIST is not enabled. MICROBIT_BLE_WHITELIST usually enabled except for OPEN security.

To see the full name all the time, use "No Pairing Required" (SECURITY_MODE_ENCRYPTION_OPEN_LINK) or set MICROBIT_BLE_WHITELIST to zero in the project settings.

The V1 Generic Access service Device Name characteristic has the WRITE property. Could this be the difference?  Any value written will be lost as soon as micro:bit is reset, so I'm not sure why WRITE is suitable. As I understand it, WRITE access could be used by the client to ask a peripheral to use a different name, but it would have to be saved in flash to be useful, and micro:bit does not do this.

For V2 this excludes WRITE
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/bluetooth/MicroBitBLEManager.cpp#L272

For V1 these enable WRITE
https://github.com/lancaster-university/nRF51822/blob/master/source/btle/btle_gap.cpp#L43
https://github.com/lancaster-university/nRF51822/blob/master/source/nRF5xGap.cpp#L538

V2 gets the full [xxxxx] name here for normal mode or pairing mode (enableBonding == pairing mode):
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/bluetooth/MicroBitBLEManager.cpp#L264

V1 gets it here for normal mode
https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitBLEManager.cpp#L302
and here for pairing mode
https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitBLEManager.cpp#L649

ukBaz, 2021-02-13:
Thanks @martinwork for taking a look at this.

Your information was correct that the difference was that the V1 micro:bit was in `No Pairing Required` and the V2 micro:bit was in `Pairing Mode`. When both devices are in the same mode, they behave the same.

Apologies for not noticing this before.

closed 2021-02-13 by ukBaz