lancaster-university/codal-microbit-v2 #79: uBit.serial.readUntil with newline delimiter sees regular extra blank lines

martinwork, 2021-02-08:
[serial-echo-readline.zip](https://github.com/lancaster-university/codal-microbit-v2/files/5945265/serial-echo-readline.zip)

The zip has a C++ sample with compiled hex files, an equivalent MakeCode project, and a python client to send lines and check the echo. The micro:bit stores the first line it receives after reset to compare with subsequent lines. It displays E if it sees an empty line and X if a line is different from the first. 

Extra blank lines occur regularly, with frequency depending on the micro:bit buffer size and the length of the lines. Every 255 lines with the samples in the zip (about 14s). More often with shorter text.

The problem occurs in the default SYNC_SLEEP mode, but not in SYNC_SPINWAIT mode. It is also present with V1 + DAL. It seemed exactly the same when I used redirection to P0/P1 with a USB to serial cable.

NB: Having a paired MakeCode window open interferes with the communication between micro:bit and the python program.

finneyj, 2021-03-08:
I think this is fixed in https://github.com/lancaster-university/codal-core/commit/4fa8275e2069434f83cdf967098ddd53644abf03

Can you please confirm @martinwork ?

martinwork, 2021-03-09:
Yes, this is fixed. Thanks @finneyj.

closed 2021-03-09 by martinwork

finneyj, 2021-03-16:
Fixed in v0.2.25