lancaster-university/codal-microbit-v2 #80: Serial - testing overlapped sending and receiving

martinwork, 2021-02-10:
[serial-repl.zip](https://github.com/lancaster-university/codal-microbit-v2/files/5957644/serial-repl.zip)

micro:bit sees errors in incoming data, if it is sending at the same time. In my tests, V2 copes much better than V1, but V2 stops responding when it fails. With the current text in the python client in the zip (31 chars), V2 sees and echoes a double character, just before it stops responding, so it doesn't look like dropped bytes. Very short lines seem not to be a problem.

For each test, flash the hex or reset micro:bit, then start the python client.

The micro:bit attempts to echo incoming serial data, while parsing lines separated by newline characters. It stores the first incoming line to compare with all the following lines, and displays a letter if is sees an error:
- R = serial.read failed
- E = an empty line was detected
- X = an incoming line doesn't match the first line

With the intention of keeping things simple, micro:bit checks the rx buffered count and the tx buffer free space, and uses ASYNC mode read and send. The buffers are large (254) and the python client waits for each line's echo, so buffer overflow cannot happen.

microbit-carlos, 2021-02-19:
@finneyj there are some accessories that maintain a longstanding serial connection to the micro:bit constantly sending and receiving data and this issue is affecting their operation. Could you have a look at this when you have the chance?

finneyj, 2021-03-06:
Thanks for the excellent repro case @martinwork @microbit-carlos 

I took a look at this last night. The lock up was caused by rx_cnt in this piece of code going negative, and resulting in the loop then walking through all the RAM after the dmaBuffer until it runs out of address space, then causing a hardfault:
https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Serial.cpp#L277

I've made the code resilient to these conditions, which fixes the lock up. I still see occasional repeated characters in your test case though. Very rare, but still there. It seems to be coming from that same block of code. If I take the code out entirely, I get a nice stable test. I left it running for a few hours (1000000 iterations) and only saw this reported, which could easily just be something timing related in the test harness: 

[image]

I don't really want to leave that code unused though. It's in there to protect against lost RXDRDY interrupts. This shouldn't happen on micro:bit at the moment as I've already set the UART to its own very high priority... But if we ever start using it with the Bluetooth stack, we'll start to see a repeat of delayed-characters bug.

I think this is all coming about because the NRF52 very occasionally reports an ENDRX event (that signifies the DMA buffer is full) before the RXRDY even that signifies that the last byte has been written... I'll keep digging.

finneyj, 2021-03-08:
I think this is fixed in thispatch @martinwork :
https://github.com/lancaster-university/codal-nrf52/commit/031d9358e72a143e4ad9bc7cdfd2f9ba04a5cb6d

Would you be happy to run through your test harness too, and close off this issue if it looks ok?

martinwork, 2021-03-09:
Thanks @finneyj. The main thing is that it doesn't crash now. I've run the 10000 iterations a few times without seeing an error.

closed 2021-03-09 by martinwork

finneyj, 2021-03-16:
Fixed in v0.2.25