lancaster-university/codal-microbit-v2 #85: BLE client mode support

wldevries, 2021-03-05:
Is there any chance we can get BLE client mode support? The codal-microbit-nrf5sdk repository includes both the S113 and the S140 Nordic SoftDevice which gives me hope, but there are no configuration options to choose and modules like `nrf_ble_scan.h` are missing.

My goal was to build blocks for MakeCode that allows me to connect to BLE enabled Lego Powered Up hubs to control Lego motors and such.

microbit-mark, 2021-03-08:
Currently, micro:bit CODAL uses SoftDevice S113 which is peripheral-mode only and was implemented in favour of S140 to save flash memory at compile time. 

If you want to use S140 to use central/client mode, you could revert this PR in your own fork of the repository https://github.com/lancaster-university/codal-microbit-v2/pull/3

You can find build instructions in https://github.com/lancaster-university/microbit-v2-samples so you could update the configuration to use your fork of CODAL.

wldevries, 2021-03-08:
Thanks! I read somewhere that extensions are built against a prebuilt CODAL/DAL on MakeCode. Does that mean it is not possible to submit extensions that are based on a modified CODAL?

I have seen some extensions for MakeCode made in C++, but I can't find any guides so I guess it's unsupported.

finneyj, 2021-03-08:
Hi @wldevries 

That's kind of true. :)

MakeCode does link against prebuilt CODAL builds, but it is also possible to configure your own build through CONFIG options to enable/disable features. You can also inject c++ into your extension if necessary which can build on CODAL.

So the way to add new low level features is to develop a minimum, reusable patch to CODAL that adds any "plumbing" you need. We can then merge that into the main branch, and you can make use of that in your MakeCode extension. From there you can surface your extension using Typescript, Python and Blocks...

wldevries, 2021-03-08:
It's great that this would be possible in theory, but I'm way out of my depth here :-D I'm a proficient .NET developer, but the custom build systems and intricacies of embedded hardware are too much I'm afraid.

finneyj, 2021-03-08:
Cool... I've written many a good project in C#! One of my favourite languages in fact.

Hmm... I don't think I'll have time to look at it any time soon I'm afraid... University staff are really swamped at the moment. I could ask around to see if any of my students fancies a look at it if you like ? A micro:bit interfacing to lego kits would be cool...

cymdiaz, 2022-04-09:
Greetings.  It is now April 2022. 

May I inquire as to the status of this issue?  I would love to be able to have the Microbit V2 act as a client as well.  I also don't know how to  "revert this PR in your own fork". 

I am currently using an ESP32 board as a client, but my boss would love if we could use the Micro:bit V2.   Don't need the blocks, but do need the scan / pair functionality.

Thanks for your time and advice.