lancaster-university/codal-microbit-v2 #86: Event.timestamp resolution?

godsic, 2021-03-08:
I was playing with HC-SR04 sensor on v2 hardware lately and noticed quite a bit of jitter in pulse width measurement of roughly 10-30 us. 

For v1 runtime/hardware it was stated in the [microbit-docs](https://lancaster-university.github.io/microbit-docs/ubit/io/)
```markdown

**Note**

In the MICROBIT_PIN_EVENT_ON_PULSE mode, the smallest pulse that was reliably detected was 85us, around 5khz. 
If more precision is required, please use the InterruptIn class supplied by ARM mbed.
```

So I was wondering does it still hold and if so, what are the alternatives to `InterruptIn` in v2 runtime?

finneyj, 2021-03-08:
Hi @godsic 

10uS - 30uS must be pretty close to the tolerance of a HC-SR04 isn't it? 10uS translates to about a 2mm-5mm range if I recall correctly? 

The data sheet here implies that's pretty much the best you can get with that sensor : https://www.electroschematics.com/hc-sr04-datasheet/

godsic, 2021-03-09:
Hi @finneyj, thanks for pointing out. Yes, the jitter is of the same order as HC-SR04 resolution. 
I am, nonetheless, curious was the events overhead reduced in v2 for improved pulse width measurements?

finneyj, 2021-03-09:
Yes, v2 is definitely more efficient than v1 for this @godsic . The pulse measurement API in particular should be pretty accurate I think. 

The events overhead is lower too, but this would only affect latency of the pulse measurement, not the accuracy of the pulse width measurement i.e. it will define the shortest detectable pulse, not the accuracy of the pulse width measurement itself... if you understand what I mean?

closed 2021-03-10 by godsic