lancaster-university/codal-microbit-v2 #92: After redirecting serial to P0 and P1, it can't be redirected back to normal USB

martinwork, 2021-03-25:
It looks like the usbTx/Rx pins' names are changed by the first redirection.

Using the test program below with some trace added to NRF52Serial::configurePins, and two tera term windows for normal USB and a USB/Serial converter cable, all the serial output after the redirection to P0&1 appears on the USB/Serial converter and the tx/rx pins at addresses 0x200028F8 0x20002910 are changed from 6 & 40 to 2 & 3.

Normal USB:
```
configurePins 0x200028F8 0x20002910 with names 6 40
configurePins PSEL 6 40
relocate vtor to 0x0 -> 0x20002200 0x20002200
test: send to usbTx/test: usbTx/Rx 0x200028F8 0x20002910 names: 6 40
Rx BEFORE
test: P0/1 0x20002628 0x20002640 names: 2 3
```

USBSerial converter:
```
configurePins 0x20002628 0x20002640 with names 2 3
configurePins PSEL 2 3
test: send to P0/1
test: usbTx/Rx 0x200028F8 0x20002910 names: 2 3
configurePins 0x200028F8 0x20002910 with names 2 3
configurePins PSEL 2 3
test: send to usbTx/Rx AFTER

```

[serial-redirect-micropython-microbit-v2-issues-70.zip](https://github.com/lancaster-university/codal-microbit-v2/files/6206432/serial-redirect-micropython-microbit-v2-issues-70.zip)

```
int NRF52Serial::configurePins(Pin& tx, Pin& rx)
{
    DMESG( "configurePins %x %x with names %d %d", (unsigned int) &tx, (unsigned int) &rx, (int) tx.name, (int) rx.name);

    this->tx = tx;
    this->rx = rx;
    nrf_uarte_txrx_pins_set(p_uarte_, tx.name, rx.name);
      
    DMESG( "configurePins PSEL %d %d", (int) p_uarte_->PSEL.TXD, (int) p_uarte_->PSEL.RXD);
    return DEVICE_OK;
}

```

finneyj, 2021-03-25:
Thanks @martinwork.

errr.... That's really weird. You can't reassign c++ references, which is what the code here seems to be explicitly trying to do. The rx/tx pins in the base class should either be pointers or simply not there at all IMHO...

martinwork, 2021-03-25:
It seems the referenced objects are being assigned. The references don't seem to be used for anything.

finneyj, 2021-03-25:
Indeed, which is why I thought we should perhaps remove them. They may be on other targets though, so for now, suggest we simply delete those two erroneous lines of code, which I think will fix it.

martinwork, 2021-03-26:
Would you like me to make a PR that deletes the assignments from configurePins?
I guess there aren't any targets subclassing NRF52Serial?

finneyj, 2021-03-26:
Hi @martinwork. If you have time to remove the lines and do a quick test that this resolves the problem then yes, I'd very much welcome a PR!

There's nothing that subclasses NRF52Serial, so should there shouldn't be any issues there.

martinwork, 2021-03-26:
Sorry, I forgot about the testing!

I've created PR https://github.com/lancaster-university/codal-nrf52/pull/28

With those 2 lines removed, redirection can work repeatedly, and my original test works smoothly.

But if I remove the sleeps I added to avoid interleaved send() and DMESG, the sent text appears on the wrong USB and some junk creeps in.

The new test below works as expected on V1, but with V2 the usbTx/Rx connection sees the following, while the P0/1 connection sees a single o circumflex.
```
`efore
during
after
`efore
during
after
`efore
during
after
`efore
during
after
–¶+during
after
–¶+during
after
–¶+during
after
```

```
    while (1)
    {
        uBit.serial.send("before\n");

#if MICROBIT_CODAL
        uBit.serial.redirect( uBit.io.P0, uBit.io.P1);
#else
        uBit.serial.redirect( MICROBIT_PIN_P0, MICROBIT_PIN_P1);
#endif

        uBit.serial.send("during\n");

#if MICROBIT_CODAL
        uBit.serial.redirect( uBit.io.usbTx, uBit.io.usbRx);
#else
        uBit.serial.redirect( USBTX, USBRX);
#endif

        uBit.serial.send("after\n");

        uBit.sleep(5000);
    }

```

finneyj, 2021-03-26:
Thanks @martinwork. 

Makes sense - v2 has a proper output buffer, so the send() operations will return once the data is queued for Tx, which will be before the bits have actually hit the wire.

The rogue characters are likely the result of either partial byte transmissions or one of the lines floating once the UART is disconnected from the pins...

We *could* wait the the Tx buffer to empty before completing the redirect, but that feels a bit invasive. We might be better leaving that to the user to decide...  You'd just need to do something like:

```
while(uBit.serial.txBufferedSize());
```

I guess an explicit flush() method or optional parameter to redirect that waits for the tx might be helpful, but not sure it's really necessary?

martinwork, 2021-03-29:
@finneyj  After a lot of thought about redirect, and looking in MakeCode and MicroPython for APIs to check the buffered size. I eventually realised that V2 send(SYNC_SLEEP) is not behaving correctly.

send(SYNC_SLEEP) is supposed to sleep until all bytes have been sent, so txBufferedSize() will be zero.

In the V1 code, send calls setTxInterrupt repeatedly to copy bytes to the tx buffer and start sending, and waits for EVT_TX_EMPTY before checking bytesWritten >= bufferLen to terminate the loop.

In the V2 code, send calls setTxInterrupt just once. setTxInterrupt loops on the string, but waits for EVT_TX_EMPTY only while it has more string that won't fit in the tx buffer. With short strings, it doesn't wait at all.

V1 looks like it could return before the last character has finished sending, but I guess that won't actually happen.

Both redirect()s look like they intend to deal with bytes in the tx buffer. That would need testing with ASYNC mode.

finneyj, 2021-03-29:
thanks @martinwork 

Should it though? Sleeping until all bytes have hit the wire would greatly affect parallelism... a serial send on most platforms would block until all data is queued. e.g. that would be semantics of unix/windows... etc. hence also python and JS on those platforms. One would wonder the purpose of having a transmission buffer is we don't use it :)

V1 did wait for data n the wire, as it didn't have enough RAM to do much else...

When you say you looked into MakeCode and micropython APIs, do you mean V! behaviour or documented API behaviour?

martinwork, 2021-03-29:
I was just going on https://lancaster-university.github.io/microbit-docs/ubit/serial/, where it says
"SYNC_SLEEP - bytes are copied into the txBuff and the fiber sleeps until all bytes have been sent.", and the realisation that V1 does wait but V2 doesn't, which is causing a problem for redirect. 

`V1 did wait for data on the wire, as it didn't have enough RAM to do much else...
`
I think V2 matched V1 until https://github.com/lancaster-university/codal-core/commit/e6b73051ed273a288536b523227915d24be01b79, and V1 could be changed to match V2's new behaviour, but then neither would match the documentation.

This issue was originally prompted by python failing to switch back to the REPL after a redirection. It will not be good if python switches back to the REPL with buffered data that was supposed to have been sent to the edge connector, but I don't know if python's uart.write depends on serial.send to be synchronous.

I couldn't find MakeCode or MicroPython APIs to check the tx buffered size.

Simple usage wouldn't expect `send(before); redirect; send(after);` to send beforeafter after the redirection. So maybe redirect should wait for an empty buffer.

I'm struggling to think of other examples that would be thrown by the V1/V2 difference:
- code that relies on the fact that serial.send sleeps to allow other fibers a look in?
- maybe unexpected timing differences when the buffer is large?

martinwork, 2021-03-29:
> Yes, fully agree on reflection that redirect() should flush the buffer.

Do you mean flush, or wait until empty?

> We could just pop in a CONFIG option in the SYNC pathway of send()

SYNC_SLEEPUNTILBUFFERED mode?

I think V2 SYNC_SLEEP/SPINWAIT should match V1 by default.

finneyj, 2021-03-29:
Sorry @martinwork, yes, I meant wait until empty. I think we should do this. 

e.g. even if the buffer was filled with send(ASYNC), redirect() should wait for the buffer to flush out before moving the pins. Technically,  waiting until the buffer is empty may still not be enough, as at the point the buffer becomes empty, the data will simply have been moved into the NRF52 hardware TX buffer, and still not been physically modulated onto the wire. We would need to wait for the EVENTS_ENDTX hardware event that is generated when UARTE hardware becomes idle...

I'm not quite so convinced about the send(SYNC) behaviour. I totally see where you're coming from for compatibility though. But changing the major revision of a device is precisely the right time to make these sorts of changes to improve performance...

It could also be equally argued that without the V2 semantics, it's impossible to efficiently maintain line speed transmissions of the serial port... and really hammers performance and lower baud rates. These each have their own associated problems too!

If send(SYNC) maintains its current behaviour, then the user can use existing APIs to wait for the transmissions to complete when they need to. If we change to the old V1 behaviour, the users have no choice...

finneyj, 2021-03-29:
I'm happy for a CONFIG options that enables the two different semantics. We can then perhaps chat to Carlos and Jonny about defaults for micro:bit, and set that default in the codal-microbit-v2 config.json ?

martinwork, 2021-04-04:
Thinking about it, if wait-until-buffered is OK maybe it's not worth complicating the code?

I've made PR https://github.com/lancaster-university/codal-nrf52/pull/31, to spin in configurePins until the TX buffer has been sent. This includes https://github.com/lancaster-university/codal-nrf52/pull/28.

This could wait for a long time with a full-size buffer and low baud. Do we need to include a sleep option? By adding an optional mode parameter to Serial::redirect?

microbit-mark, 2021-04-14:
Checking in on the status of this one as it's still causing issues for the person that raised https://github.com/microbit-foundation/micropython-microbit-v2/issues/70 I believe it affects use of the TI calculator https://education.ti.com/en/teachers/microbit

martinwork, 2021-04-19:
@finneyj I think https://github.com/lancaster-university/codal-nrf52/pull/31 works, unless we need to do more for the points in https://github.com/lancaster-university/codal-microbit-v2/issues/92#issuecomment-813033053.

martinwork, 2021-06-07:
Closing as https://github.com/lancaster-university/codal-nrf52/pull/31 has been merged.

closed 2021-06-07 by martinwork