lancaster-university/codal-microbit-v2 #93: MicroBitPartialFlashingService in MakeCode Beta is returning zero for region start and end addresses

martinwork, 2021-03-31:
I USB flashed a new hex from MakeCode Beta, paired and attempted flash with both release and beta iOS apps.

When the PF service is available, the iOS app checks the regions before starting to flash to compare the DFU portion of the hex with the values of the software in micro:bit, to avoid mismatches in softdevice or bootloader.

But the region addresses are zero, so flashing to V2 is not possible in the app with new hex files.

This is a bit rushed, so I could be wrong, but it seems OK with MakeCode live.

@microbit-sam 
@abchatra

martinwork, 2021-03-31:
MicroBitMemoryMap used to, and should, always create the basic regions, but now only does so if it finds MakeCode or Python hashes. It seems it might not be finding the MakeCode hashes.

martinwork, 2021-04-01:
@abchatra I'm pretty sure this is purely a CODAL issue. I tagged you to flag up that if the current MakeCode beta goes live, the iOS app will stop being able to flash MakeCode hexes.

microbit-sam, 2021-04-01:
This is probably from the PR that added uPy partial flashing, I don't think I touched any of the MakeCode related code but it might be getting cleared when the uPy magic isn't detected. Will take a look

abchatra, 2021-04-01:
@martinwork is this broken with /stable too? What version of codal this repros?

martinwork, 2021-04-01:
stable seems OK. I believe the problem arose in v0.2.25.

martinwork, 2021-04-01:
Oh... but stable and beta are both 0.2.25.
A beta hex has 32 byte records, while stable has 16.

martinwork, 2021-04-01:
@abchatra @mmoskal In my file saved from beta, the MakeCode magic is at 44c00. It is expected to be on a page boundary.

@microbit-sam That's why MemoryMap isn't finding it and generating the regions. But the iOS app still wants to check the region addresses when it's a plain C++ hex.

microbit-sam, 2021-04-06:
Thanks Martin, will be simple enough to scan entire pages. This is also going to affect V1 microbit-dal, right?

martinwork, 2021-04-06:
@microbit-sam We agreed a while back that the magic will be on a page boundary, so for V2, it should be at 45000, not 44c00, V1. DAL can continue to use an 0x400 boundary. The current apps don't expect V2 magic on a 0x400 boundary either, so updating the V2 MemoryMap code  won' t fix it.

@abchatra @mmoskal Is this right, or are we changing V2 to 0x400 boundary now?

@microbit-sam The released iOS app checks the region addresses if the PF service is present. This was introduced when we switched V2 softdevice, to avoid bricking a micro:bit with S140. The app checks this for V1 and V2, simply if the PF service is present. But the new V2 MemoryMap does not generate the regions unless it finds MakeCode or Python hashes, so a C++ hex in micro:bit will supply zero region addresses. This change will stop the released iOS app from being able to flash over a C++ hex which has the PF service. Although I have just added a check to the development version to ignore junk addresses, I think we should stick with the PF service supplying useful addresses for softdevice region etc. even when it doesn't find the editor hashes.

microbit-carlos, 2022-05-05:
@martinwork is this still an issue in CODAL/MakeCode?

martinwork, 2022-05-05:
The MakeCode issue was fixed, but I don't think the PF service has been changed.

It stopped generating regions for C++ programs on 15 Mar 2021 (https://github.com/lancaster-university/codal-microbit-v2/commit/5783660ee7c4fbfe9ccd855cc102353ba16860fb#diff-0aeb91fe81d6c38692780252a1d85d9fd9d1a9cda304535b1aa369b075616dffL48) and there are no later changes.

The reasons for this being a potential problem (e.g. detecting S140) are now perhaps in the past.

I still think that "MicroBitMemoryMap used to, and should, always create the basic regions, but now only does so if it finds MakeCode or Python hashes.", but this is only a problem if the PF service is included in a C++ hex. The iOS app was updated to cope with junk values.