lancaster-university/codal-microbit-v2 #97: Feature: Turn microphone LED off when not in use

microbit-mark, 2021-04-29:
> From support ticket #44785

When the microphone is used as part of a program, the LED is lit to signify that it is listening and it remains on for that program.

It may be the case that someone wants to only use the microphone in part of a script. In this Python example, I am only using the microphone functionality when button A is pressed. When it is not pressed, I would expect the microphone to be disabled and as such the LED to turn off.

``` Python

from microbit import *

soundGraph = Image("11111:"
                    "11111:"
                    "11111:"
                    "11111:"
                    "11111")

while True:
    if button_a.is_pressed():
        display.show(soundGraph * microphone.sound_level())
    else:
        display.show(Image.HAPPY)
```

ukBaz, 2021-05-04:
I have also experienced this issue with MakeCode. 
My use case was to sample the sound level fairly infrequently and so was looking at this to clearly communicate with users when the microphone is listening/active. 
I am also assuming there is some power savings to be had by turning this off when not required.

Wanderlustcuckoo, 2021-05-06:
This is something I have experienced.  I've got a program that uses the microphone in one case, but the user can move to other functions that don't use it and the microphone LED stays on all the time.  It actually defeats the purpose of the LED if it comes on when people are not being recorded.

microbit-mark, 2021-06-22:
People are likely going to want to log sound levels as part of data logging and the mic LED being permanently on will be a drain on battery power. @finneyj This seems like a good feature to fold into the next round of work on the sound APIs if it will benefit data logging too.

finneyj, 2021-06-22:
Fully agree. We just a policy for what "not in use" means. We need to remember that the only way to turn off the LED is to turn off the microphone entirely - at which point it will likely take a little while to turn back on again and produce an accurate result. Choosing that period of time is therefore important. Also, for applications using streamed data, the microphone should stay enabled until they disconnect from the input stream.

So as a start point, a policy might look a bit like

  - microphone is disabled by default
  - microphone is enable for, say, 200ms every time the volume level is polled, then is automatically disabled.
  - microphone is enabled whenever a streaming interface is connected to it (FFT, audio recorder, ML recogniser), and remains enabled until all such interfaces are disconnected.

ukBaz, 2021-06-22:
Is it possible to have a way to manually turn off the microphone? For me the issue was that querying the volume automatically powered the mic but there was then no way for me to turn it off.
Rather than come up with a magic number (e.g. 200ms), an API to power the mic off would be acceptable for my use case. I want to take several readings in a short period then switch off for a long period. Currently there is no API for me to power off the mic.

finneyj, 2021-06-22:
For sure - manual API is totally fine. You can do this in CODAL simply via:

```
uBit.io.runmic.setDigitalValue(0);
```

I don't think that's surfaced in the higher level languages though. For kids it's probably a good idea to auto disable after the microphone goes unused for a period of time though - just to lower their barrier to entry. I don't these things are mutually exclusive though - we can do both. Would that cause a problem for you if you also had the manual function @ukBaz?

ukBaz, 2021-06-22:
> I don't think that's surfaced in the higher level languages though.

I think that is the root of my problem.

> I don't these things are mutually exclusive though - we can do both. Would that cause a problem for you if you also had the manual function?

I don't think it would be a problem. My only concern would be as you said "it will likely take a little while to turn back on again and produce an accurate result" so if it turns off too quickly...

> For kids it's probably a good idea to auto disable after the microphone goes unused for a period of time though - just to lower their barrier to entry. 

I would have thought the current setup is the lowest barrier to entry. If you care about turning off then you are probably at a level to be able to use a manual off. So if I was to express a preference, my vote would be that the auto-off is unnecessary but I can see the pros and cons of both approaches.

Thanks for your discussion on this matter. Very much appreciated.

finneyj, 2021-06-22:
Yes, we'll need to measure and account for any power on delay... that should be abstracted out in a call to read the current volume I think. (i.e. if that call resulted in the device being powered on, it should wait until it's ready before returning a result).

Just having an explicit call to disable the microphone could be an option - one for us to discuss with the MakeCode and micropython teams I think. 

Also @JoshuaAHill has been kindly looking to clean up the audio input pipeline and standardize it's use a bit across platforms.  Perhaps we could squeeze this feature request into his upcoming PR.

microbit-carlos, 2022-05-05:
@JohnVidler has this been implemented already in the upcoming release?

microbit-carlos, 2022-05-13:
Released with v0.2.37 🎉 
@JohnVidler please reopen if that's not the case.

closed 2022-05-13 by microbit-carlos