lancaster-university/codal-microbit-v2 #102: Program freezes when accelerometer is used on V2

microbit-mark, 2021-05-10:
When using the accelerometer in a loop the program will freeze after a while. In MakeCode this can be 5-8 minutes, in MicroPython sooner. Does not reproduce on V1. Here's a short reproducer in MakeCode:

https://makecode.microbit.org/_1CdXCbXCRFTD

[image]

Also reported as affecting this project on the website https://microbit.org/projects/make-it-code-it/microbit-pet/

Ziriax, 2021-05-10:
Might be related to #98 ?

microbit-mark, 2021-05-10:
Doesn't seem to trigger in https://python-editor-2-2-0-beta-5.microbit.org/ so it seems like a recently introduced bug @martinwork @finneyj 

``` Python
# Add your Python code here. E.g.
from microbit import *
import music

accelX = 0
accelY = 0

while True:
    display.set_pixel(2, 2, 0)
    accelX = accelerometer.get_x()
    accelY = accelerometer.get_y()
    music.pitch(262, 50)
    display.set_pixel(2, 2, 9)
    sleep(1000)
```

finneyj, 2021-05-10:
Thanks @microbit-mark.

Following up on #98 I had left a native C++ program running last night that accessed the accelerometer. It was working just fine 12 hours later... so I think there may be something more afoot here.

finneyj, 2021-05-10:
Thanks for sharing the code @microbit-mark. Does the bug still reproduce if you don't use the music API?

microbit-mark, 2021-05-10:
> Thanks for sharing the code @microbit-mark. Does the bug still reproduce if you don't use the music API?

It does. If you remove the accelerometer code, it does not trigger an issue

microbit-carlos, 2021-05-10:
This is the CODAL version for the MicroPython beta.5 release: v0.2.25
https://github.com/microbit-foundation/micropython-microbit-v2/blob/v2.0.0-beta.5/src/codal_app/codal.json#L5

finneyj, 2021-05-10:
err... hang on. v0.2.25 is the latest release!

So does if we can't reproduce on MicroPython beta5, where can we reproduce?

microbit-carlos, 2021-05-10:
Looks like MakeCode is in the same version:  
[image]

https://github.com/microsoft/pxt-microbit/blob/v3.0.68/pxtarget.json#L161-L166

@microbit-mark can you check what versions of MakeCode and MicroPython this can be reproduced?

martinwork, 2021-05-10:
This also freezes https://makecode.microbit.org/_E6EHi30DCP0Y
[image]

finneyj, 2021-05-10:
Interestingly, this one doesn't:

[image]

finneyj, 2021-05-11:
Just for tracking:

I've spent 30 mins or so looking at this in a debugger, and the I2C hardware seems to have got itself into an inconsistent state. When the device locks up, the NRF52 is just finishing the first half of an I2C repeated write/read transaction on the accelerometer (a very typical operation). The I2C registers look like this, while the software is waiting for the hardware to go into a SUSPEND mode:

```
I2C: WAIT_FOR_STOP():
  EVENTS_STOPPED: 0
  EVENTS_ERROR: 0 [0]
  EVENTS_SUSPENDED: 0
  EVENTS_RXSTARTED: 0
  EVENTS_TXSTARTED: 1
  EVENTS_LASTRX: 0
  EVENTS_LASTTX: 1
  TXD.MAXCNT: 1
  SHORTS: 256
  ENABLE: 6
  TXD.AMOUNT: 1
```

This should never happen, as the SHORTS configuration links the LASTTX event to the SUSPEND event in hardware. But this doesn't seem to be happening for some reason...

martinwork, 2021-05-11:
At this stage, this is expected, but magnetometer suffers the same problem.

Could we side-step by implementing readRegister like this?
https://infocenter.nordicsemi.com/topic/ps_nrf52833/twim.html?cp=4_1_0_5_28#concept_ebf_cwp_xr__fig

finneyj, 2021-05-11:
OK... So this is a curious one. I think we're looking at a minor silicon bug here on the NRF52.

The NRF52I2C driver configures the I2C peripheral to place a hardware SHORT between the LAST_TX event and either the SUSPEND / STOP task (depending on whether or not a repeated start condition is being used).

It looks a **lot** like this SHORT sometimes just doesn't trigger. The likelihood of this happening seems to be broadly related to the level of background IRQ level processing. The more IRQ, the more likely the issue is to appear. 

I've worked up a patch that detects these conditions, and triggers the relevant task in software should it occur. 
https://github.com/lancaster-university/codal-nrf52/pull/36

This seems to resolve the excellent yet epileptic C++ repro case kindly provided by @dpgeorge here: 
https://github.com/microbit-foundation/micropython-microbit-v2/issues/73#issuecomment-837964666

Unless anyone can see an issue with the PR, I suggest we merge that in and see if it addresses other similar bug reports....

microbit-mark, 2021-06-09:
This appears to be fixed in CODAL v0.2.26 as per comment https://github.com/microsoft/pxt-microbit/issues/4105#issuecomment-848684339 I can no longer reproduce using the initial code in https://github.com/lancaster-university/codal-microbit-v2/issues/102#issue-884315574 can we close?

finneyj, 2021-06-09:
I believe so - we've not had any other reports of this issue. Let's reopen this ticket if it ever resurfaces...

closed 2021-06-09 by finneyj