lancaster-university/codal-microbit-v2 #105: Bluetooth uart only send 20 bytes (the MTU) despite SYNC_SLEEP being set

EmilyBjartskular, 2021-05-15:
Below is a stripped down version of my code. In uartSend `bytes` correctly returns the actual length of msg.
For reading the data I have tried the android apps `nRF Connect` and `micro:bit UART`.

```c
#include "MicroBit.h"

void uartSend(ManagedString s);

MicroBit uBit;
MicroBitUARTService* uart;

ManagedString msg =
    R"(abcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyz)";

void uartSend(ManagedString s) {
  uint8_t bytes = uart->send(s, SYNC_SLEEP);
}

void onClickB(MicroBitEvent) {
  uartSend(msg);
}

int main() {
  uBit.init();

  ManagedString name = "Abc 123";
  ManagedString sn = uBit.getSerial();

  uBit.bleManager.init(name, sn, uBit.messageBus, uBit.storage, true);

  uBit.sleep(1000);

  uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK,
                         onClickB);

  new MicroBitTemperatureService(*uBit.ble, uBit.thermometer);

  uart = new MicroBitUARTService(*uBit.ble, 90, 90);

  if (uBit.bleManager.getCurrentMode() != MICROBIT_MODE_PAIRING) {
    uBit.bleManager.pairingMode(uBit.display, uBit.buttonA);
  }

  release_fiber();
}
```

martinwork, 2021-05-16:
Related: https://github.com/lancaster-university/microbit-dal/issues/173

Sorry @EmilyBjartskular, you are right. V1 has the same problem.

Note that main() should not call bleManager.init or bleManager.pairingMode. uBit.init does that, so long as the codal.json has the correct settings to enable BLE.

I'll post my test sample when I fix the buffering in send().

martinwork, 2021-05-17:
@EmilyBjartskular I have made a PR. Here is my test and codal.json.

```
#include "MicroBit.h"
#include "MicroBitUARTService.h"

MicroBit uBit;
MicroBitUARTService* uart;

//ManagedString msg = R"(abcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyz)";
ManagedString alpha = "abcdefghijklmnopqrstuvwxyz\n";
ManagedString msg = "  0" + alpha + "  1" + alpha + "  2" + alpha + "  3" + alpha + "  4" + alpha + "  5" + alpha + "  6" + alpha + "  7" + alpha + "  8" + alpha + "  9" + alpha;

int uartSend(ManagedString s)
{
  // default mode is SYNC_SLEEP
  // return value is int, to allow for error codes
  int bytes = uart->send(s);
  DMESG( "uartSend bytes = %d", bytes);
  return bytes;
}

int uartSendAsync(ManagedString s)
{
    int length = s.length();
    int sent = 0;

    while ( sent < length)
    {
        int bytes = uart->send( s.substring( sent, length - sent), ASYNC);
        if ( bytes > 0)
        {
            DMESG( "uartSendAsync bytes = %d", bytes);
        }
        if ( bytes < 0) return bytes;
        sent += bytes;
    }
    return sent;
}

void onClickB(MicroBitEvent) 
{
    int sent;
    sent = uartSend( msg);
    DMESG( "onClickB sent = %d", sent);
    sent = uartSend("\n");
    DMESG( "onClickB sent = %d", sent);
}

void onClickA(MicroBitEvent) 
{
    int sent;
    sent = uartSendAsync(msg);
    DMESG( "onClickA sent = %d", sent);
    sent = uartSendAsync("\n");
    DMESG( "onClickA sent = %d", sent);
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    new MicroBitTemperatureService(*uBit.ble, uBit.thermometer);
    uart = new MicroBitUARTService(*uBit.ble, 90, 90);

    uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK, onClickB);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onClickA);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

```
{
    "target": {
        "name": "codal-microbit-v2",
        "url": "https://github.com/lancaster-university/codal-microbit-v2",
        "branch": "master",
        "type": "git",
        "test_ignore": true,
        "dev": true
    } ,
    "config":{
        "DEVICE_DMESG_BUFFER_SIZE":4096,
        "DMESG_SERIAL_DEBUG":1,

        "DEVICE_BLE": 1,
        "MICROBIT_BLE_ENABLED" : 1,
        "MICROBIT_BLE_PAIRING_MODE": 1,
        "MICROBIT_BLE_DFU_SERVICE": 1,
        "MICROBIT_BLE_DEVICE_INFORMATION_SERVICE": 1,
        "MICROBIT_BLE_EVENT_SERVICE" : 1,
        "MICROBIT_BLE_PARTIAL_FLASHING" : 1,
        "MICROBIT_BLE_SECURITY_LEVEL": "SECURITY_MODE_ENCRYPTION_NO_MITM"
    }
}

```

EmilyBjartskular, 2021-05-17:
@martinwork Thank you very much! I have checked out the PR-branch and tried your test code. So far it seems to work flawlessly.

On a slightly unrelated note I have opened #107 as I have the same connection issues with your sample as with my code.

martinwork, 2021-06-07:
@finneyj PR is https://github.com/lancaster-university/codal-microbit-v2/pull/106.

closed 2021-06-07 by finneyj