lancaster-university/codal-microbit-v2 #107: Ble connection issues. bleManager.pairingMode() needs to be explicitly called

EmilyBjartskular, 2021-05-17:
> Note that main() should not call bleManager.init or bleManager.pairingMode. uBit.init does that, so long as the codal.json has the correct settings to enable BLE.

For more context see the entire chain on #105.

codal.json (note that the url/branch is not an issue)
```json5
{
  "target": {
    "name": "codal-microbit-v2",
    "info": "This target specifies the micro:bit V2, based on the nordic NRF52833",
    "url": "https://github.com/lancaster-university/codal-microbit-v2",
    "branch": "master",
//    "url": "https://github.com/martinwork/codal-microbit-v2",
//    "branch": "ble-uart-send",
    "type": "git"
  },
  "config":{
    "DEVICE_DMESG_BUFFER_SIZE": 4096,
    "DMESG_SERIAL_DEBUG": 1,

    "DEVICE_BLE": 1,
    "MICROBIT_BLE_ENABLED" : 1,
    "MICROBIT_BLE_PAIRING_MODE": 1,
    "MICROBIT_BLE_DFU_SERVICE": 1,
    "MICROBIT_BLE_DEVICE_INFORMATION_SERVICE": 1,
    "MICROBIT_BLE_EVENT_SERVICE" : 1,
    "MICROBIT_BLE_PARTIAL_FLASHING" : 1,
    "MICROBIT_BLE_SECURITY_LEVEL": "SECURITY_MODE_ENCRYPTION_NO_MITM"
  }
}
```

Regarding `bleManager.init()` etc, it was my understanding too that they should not be explicitly called, but while I could indeed remove `bleManager.init()` it seems like I still need to explicitly call `bleManager.pairingMode()` in order to connect to the micro:bit. Neither `"MICROBIT_BLE_PAIRING_MODE": 1` nor `bleManager.advertise()` (for trying to connect with previously paired device) seems to work.

`nRF Connect` (android app) shows that the micro:bit does not advertise with only "MICROBIT_BLE_PAIRING_MODE" being set.
When using `bleManager.advertise()` `nRF Connect` can see that the micro:bit is scanning but connecting does not work despite already being paired.

Explicitly calling `bleManager.pairingMode()` does allow connecting to the micro:bit, both for already paired devices and also new devices.

martinwork, 2021-05-17:
With a secure pairing mode like SECURITY_MODE_ENCRYPTION_NO_MITM, the micro:bit only advertises to paired devices, unless it is reset into pairing mode.

Pairing is easiest with the Android or iOS micro:bit app, though on Android you can pair in Settings after putting micro:bit into pairing mode.

The usual way into pairing mode is to press and release the micro:bit RESET button while holding down buttons A and B, then continue to hold A and B until the LEDs fill and then show a Bluetooth symbol. micro:bit will then show its identifying histogram and be advertising to all devices.

Having paired micro:bit with a device, that device will see micro:bit advertising in normal application mode.

It's possible to do without secure pairing by specifying "SECURITY_MODE_ENCRYPTION_OPEN_LINK" in the codal.json.

See https://github.com/martinwork/microbit-v2-samples/blob/master/source/samples/BLETest.cpp#L67

martinwork, 2022-12-30:
bleManager.pairingMode() is called [here ](https://github.com/lancaster-university/codal-microbit-v2/blob/master/model/MicroBit.cpp#L261) when A+B+RESET is used to enter pairing mode.

closed 2022-12-30 by martinwork