lancaster-university/codal-microbit-v2 #108: MicroBitLog uses innerHTML wich leads to XSS attaches

pelikhan, 2021-05-18:
The user is able to inject HTML/Javascript into the data page.

```
input.onButtonPressed(Button.AB, function() {
    flashlog.clear()
})
flashlog.setTimeStamp(FlashLogTimeStampFormat.Milliseconds)
basic.forever(function () {
    led.toggle(0, 0)
    const ax = input.acceleration(Dimension.X)
	flashlog.beginRow()
    flashlog.logData(`a.x`, `<a href="https://bing.com">${ax}</a>`)
    flashlog.endRow()
})
```

pelikhan, 2021-05-18:
Recommendation is to use DOM operation to set the value of cells instead of innerHTML.

microbit-matt-hillsdon, 2021-05-21:
This issue can be fixed by using innerText or other DOM API rather than innerHTML, but there's a related injection issue due to the use of a comment to package the data.

There's nothing preventing user text terminating the comment, outputting arbitrary HTML/JS (and potentially resuming the comment).

E.g. (in C++ syntax):

```
uBit.log.logData("xss2", "--><svg onload='alert(2)'><!--");
```

I think this will have to be addressed on the C++ side.

https://html.spec.whatwg.org/multipage/syntax.html#comments

finneyj, 2021-05-21:
Thanks @microbit-matt-hillsdon 

yep - thanks for the reminder! I always meant to prevent the user from adding `-->` into the datastream...

If you can patch up my dodgy HTML @microbit-matt-hillsdon  I'll just filter the `-->` symbol out at ingress?

pelikhan, 2021-05-21:
Why are you using a text format? You could write the data payload a hex buffer and parse it in javascript directly. In such case, you fully control the characters being injected in the HTML source.

microbit-matt-hillsdon, 2021-05-21:
I think we should do this if we can. @finneyj what's the background to the comment approach? Is it necessary that we have only a HTML "header" and no trailer?

pelikhan, 2021-05-21:
What happens when the user insert <, >, or - ? Can you guarantee there is no way to escape the comment sandbox?

pelikhan, 2021-05-21:
Writing the encoded data as a hex buffer prevents any class of javascript injection that you are exposing yourself with writing user data in a HTML comment.

finneyj, 2021-05-21:
Fully agree about HEX encoding being the preferred approach @pelikhan, and with the new version of DAPLink that's precisely the plan. That version can do this encoding for us dynamically on demand, but current (live and deployed) versions of DAPLink don't support this feature.  

We could write as a HEX buffer at source at the cost of losing 50% of available data storage (dropping effective data storage from ~120KB to ~60KB). If we can make this approach secure by controlling the data content, it would be preferable,. Otherwise, we either need to force users needing to update DAPLink to be able to the logging features, or drop storage efficiency by 50%.

I can't see why we can't guarantee that all `-->` symbols are removed. All writes go through a single, line buffered function that writes a single (`\n` terminated) row to the table. If we scan that line before writing for the `-->`  symbol, we can strip it, we can close the hole.  Can you see a reason why that wouldn't work @pelikhan?

finneyj, 2021-05-21:
@microbit-matt-hillsdon The HTML trailer proved irrelevant in testing. All browsers I tested moved the </BODY></HTML> tags within the DOM to be at end of file. Putting these tags within the 2KB header block just burned storage.

pelikhan, 2021-05-21:
> I can't see why we can't guarantee that all --> symbols are removed.

Is there a way to craft those characters in some kind of weird unicode encoding, or generate a malicious input with >,< character that forces the browser so treat the comment as closed? I'm not sure.

What i mean by encoding buffer is generate javascript code with a hex buffer in it, then decode

```
var data = "abcdadsfasdfjkasldfjasdkfljasdlfkasjfdlkjasdfkj"
// javascript to decode array goes here
```

pelikhan, 2021-05-21:
> I can't see why we can't guarantee that all --> symbols are removed. 

Is there a way to craft unicode literals or malicious entries that will treat the browser as a bogus comment and close it sooner than the -->? I don't know. The HTML parser recovers from errors, and it's hard to test for all those cases.

finneyj, 2021-05-21:
hmm, true - unicode literals could provide a route in that's difficult to catch. We'd have to restrict input further. If we were confident it's just literals, then we could restrict ampersand `&` too.  If we're more cautious then just alphanumerics. 

Alternatively, hexify at source, accept the 50% overhead and send a message that "upgrade your interface firmware to get more storage". Turn a problem into an opportunity for improvement. :)

JohnVidler, 2022-06-10:
Moved to v0.2.40 so we can close off the v0.2.39 milestone. I expect that this will be in the minor patchest for *40.

JohnVidler, 2022-06-15:
The current implementation (As seen at https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitLog.cpp#L678 ) handles the commenting-out attack, and I'd rather keep this for now.

The only way to go with this right now without extensive testing would be a kind of 'paranoid mode' where only alphanumeric ASCII characters are allowed, which would have problems with anyone trying to encode unicode into the strings.

Consequently, I'm going to bump this to 0.2.41, and give us time to correct it properly.

jaustin, 2022-06-23:
> Is there a way to craft unicode literals or malicious entries that will treat the browser as a bogus comment and close it sooner than the -->? I don't know. The HTML parser recovers from errors, and it's hard to test for all those cases.

The key outstanding issue for me is answering this question, but I'd also like to work through the downsides (if any) of limiting ourselves to alphanumerics. I'm not sure what MakeCode supports here either - @pelikhan what kind of strings does MakeCode support?

pelikhan, 2022-06-23:
Can you escape on the html page side and use the dom to populate the table?

JohnVidler, 2022-07-07:
The main way to inject via closing an HTML comment has been mitigated, but without any evidence of this being explotable with the existing code then we should leave this as it is.

JohnVidler, 2022-08-10:
Until we get a case of someone actually doing this, the existing patches seem to be sufficient, and as trying to guess attack vectors would be a bit like playing 'bug whack-a-mole', I'm going to leave this and we should create new, specific issues for any vectors found in the future.

closed 2022-08-10 by JohnVidler