lancaster-university/codal-microbit-v2 #109: MicroBitLog page empty for single column / no timestamp logging

pelikhan, 2021-05-18:
JS expects at least 2 columns for some reason. Nothing happens when logging a single entry, not timestamp.

finneyj, 2021-05-25:
Thanks @pelikhan - data logging in flash looks ok when one row is in use, so agree this looks like it's on the JS side. Matt has been busy refactoring and improving the HTML+JS part anyhow, so may get fixed on the way. 

@microbit-matt-hillsdon Let's test this when we merge your PR with updated HTML+JS is ready.

microbit-matt-hillsdon, 2021-05-25:
I've reproduced this on master and confirmed it is fixed as part of the changes on #111. I used this C++ test program to test:

```
    for (int i = 0; i < 100; ++i) {
        uBit.log.beginRow();
        uBit.log.logData("a.x", uBit.accelerometer.getX());
        uBit.log.endRow();
        uBit.sleep(1000);
    }
```