lancaster-university/codal-microbit-v2 #110: MicroBitLog header improvements

jaustin, 2021-05-19:
We'd like to make some simple changes to the micro:bit log.

## The log has two distinct 'modes'
Offline:
- A simple file to get the data out. Minimum styling (to save flash)
Online
- Loads some resources from microbit.org to augment the page (css, js)
- Initially this shouldn't add features but should style the page and if possible include more descriptive dialogs

### The changes required are
* Make the online mode exist
* Create an auto-update mechanism
* Style/tweaks to offline mode
* Fix #108 

### Optional
* Improve the 'pipeline' that creates the minified HTML/JS as described in https://github.com/lancaster-university/codal-microbit-v2/tree/master/resources/logfs
* 'space used' indicator that tells the user what percentage of the storage is left (could be on a page level...). This could be in online mode only.
 

## Autoupdate

In both modes we'd  like to use something like the following mechanism to 'auto-update' the data contents when the micro:bit de/re-enumerates
- In a hidden iFrame, periodically reload the MY_DATA.HTM file and use a mechanism (eg post messaging) to determine that the file contents of the two file are (not) the same. If contents do not match, reload ourselves to show the latest content.
- it's likely that comparing the header metadata will do a good job of this, as all writes (currently) result in the length getting incremented in the header block. However @finneyj and I discussed moving to the 'length' field becoming a page pointer and relying on the trailing 0xFFs to determine the end of the data in the relevant block. In this case, the flash page AND the last !0xFF address would be a good proxy for 'is the data different' mechanism. @finneyj could you recommend the smallest/simplest 'hash' for identifying changed data (rare/unlikely collisions are fine...)
 
## Style changes:
* heading should be "micro:bit data log" (IE title case)
* Button titles should be title case, and ellipsis on the end of the two buttons that popup dialogs (or some 'i' icon to show they're information not actions).
* horizontal spacing between the buttons (just a single space kinda works but feels like a hack, but it is only 4 bytes!)
* Vertical space between buttons and table start
* In the table: headers should be center aligned, data right aligned

## For the 'online hydrated' page
* micro:bit logo and simple branding @microbit-matt-hillsdon please do something simple for now and we'll review
* buttons on brand (black, rounded edges)

Do we want to do something more intelligent with the table too? I think it's quite effective as is and there's a benefit to online/offline looking as similar as possible.

jaustin, 2021-05-19:
@microbit-matt-hillsdon I don't appear to be able to assign you to this, but here's the HTML stuff we discussed

finneyj, 2021-05-19:
Sounds good to me.  Are we constraining this to the current 2K maximum size for the HTML/JS?

For a hash, I would suggest a simple 32-bit sum of the whole file system. We have a fast CPU, so that's probably ok for most changes, and would minimize code size.

microbit-mark, 2021-06-11:
The pop-up text when you click on a button is confusing and possibly intimidating for new users. It's also a bit frustrating as you expect the buttons to have the described action:

- What is `file://` and why is it there. Could we hide this or change it to `MY_DATA.HTM`
- Is it useful to ask people whether they want not to be prompted again? Why is this an option for clear log but not for update data?
- clear log

[image]

> "The log is cleared when you reflash your micro:bit. Your program can include code or blocks to clear the log when you choose"

Do we expect people to know what "reflash" means and how to do this? Could we link people to a webpage or KB article on this or direct them back to the editor they were using and to press download again?

Maybe something like this:

"To clear the log you need to [flash](https://microbit.org/get-started/first-steps/set-up/#transfer-from-a-computer) another program to the micro:bit. [Clearing the log in MY_DATA.HTM](#)"

We would link to the KB article which would describe the process and the blocks/APIs that you can also use.

- update data
[image]

> "latest data that changed" doesn't read well and we should try and use the present tense.

Maybe:

"Disconnect the USB cable from the micro:bit and then reconnect it to update the data in the table. Rows will be added to the bottom of the table. [Updating the log in MY_DATA.HTM](#)"

Link to the KB for more info

microbit-mark, 2021-06-11:
Another issue. When the table updates, the data is added to the bottom of the table. There is no prompt to let people know that this has happened. Could we have a visual cue that the data in the table has been updated. Maybe a button to scroll to the bottom of the table?

microbit-mark, 2021-06-11:
When I click 'Copy' the button changes to 'Copied'.  I would expect it to revert back to 'Copy' once that data is copied to the clipboard
[image]

Firefox 89 mac OS 11.2.3

microbit-matt-hillsdon, 2021-06-11:
Thanks Mark. Initial thoughts on the above feedback below.

## Pop-up dialogs

Unless we implement custom message dialogs (which we could consider doing for online mode), we're not in control of anything other than the text in the body of the standard "alert" dialog. So we have no control overt the file:// indicator shown by the browser or whether the browser decides our page is annoying the user and includes the "Don't allow" option (I suspect you get a pass for the first dialog but then the browser shows it for subsequent ones). The alerts differ in these areas between browsers. We also cannot include a clickable link in a regular alert.

We could consider raising a new issue for richer custom dialogs for online mode that could include hyperlinks etc.

This might be a good thing to work on alongside charting if we do that.

## Prompt when table updates

The page is completely reloaded when the data changes so this is a bit of a pain to achieve, as the new page doesn't know why it was loaded. I agree some kind of indicator would be useful. We could investigate using the URL or session storage to track that we have just been reloaded. Probably not enough space to do this in offline mode.

## Copy button

It's definitely intended to change back to "Copy" and this was working for me. I haven't built a fresh HTML file but it works in the last ones I saved when working on this.

jaustin, 2021-06-14:
See https://github.com/lancaster-university/codal-microbit-v2/issues/121 for discussion about autoupdate

microbit-carlos, 2022-05-13:
All the non-optionals in this ticket are done 🎉

closed 2022-05-13 by microbit-carlos