lancaster-university/codal-microbit-v2 #112: Support for NeoPixel/WS2812 LEDs?

joachimhs, 2021-05-27:
Hello, 

Is there built in support for programming NeoPixels in Codal-Microbit? I saw an issue that it was added in 0.2.20 (https://github.com/lancaster-university/codal-microbit-v2/issues/19), but I was unable to find an example on how to program these using the "MicroBit.h" library, and compilation via the Docker image found in https://github.com/lancaster-university/microbit-v2-samples.

finneyj, 2021-05-27:
Hi @joachimhs 

Yes there is, but it is a little hidden, as we enabled it for all NRF52 based devices (not just microbit). You should just be able to call either of these two function to get hardware accelerated neopixel support:

https://github.com/lancaster-university/codal-nrf52/blob/master/inc/neopixel.h#L15

The function will block your running thread until the send is complete. if you want it to run in the background, just swap the function on this line to "playAsync" instead of "play".

https://github.com/lancaster-university/codal-nrf52/blob/master/source/neopixel.cpp#L25

Hope this helps.  I'll see if I have some sample code...

joachimhs, 2021-05-28:
Thank you for your quick reply, @finneyj !

An example would be great, if you have one!

Very Best Regards,
Joachim

finneyj, 2021-05-28:
No problem @joachimhs. 

I couldn't find my test sample, but something like this should do.. I'm afraid I don't have a neopixel strip to hand, so couldn't test it. Let me know if this works! :)

```
#include "neopixel.h"

void
neopixel_test()
{
    // Length of neopixel strip
    int length = 8;

    // Colour counter variable (just used to remember what colour we last displayed)
    int c = 0;

    // Create a buffer to hold pixel data of the appropriate size
    ManagedBuffer b(length*3);

    while(1)
    {
        // Clear the buffer to zeroes. You don't need to do this, but it's often convenient.
        b.fill(0);

        // Fill the buffer with some data.
        // Neopixel typically has 3 bytes per pixel, for each primary colour
        // in GRB format (Green, then Red, then Blue).
        for (int i=0; i<length*3; i+=3)
            b[i+c] = 255;

        // Update the neopixel strip on pin P0.
        neopixel_send_buffer(uBit.io.P0, b);

        // Increment our colour counter so that we show a different colour in the next frame.
        c = (c+1) %3;

        // Wait for a second.
        uBit.sleep(1000);
    }
}
```

joachimhs, 2021-05-28:
Ah, this works like a charm! 

I am now able to finally provide NeoPixel support for my blocks-based software for programming micro-controllers that I have made for kids in Norway :) 

I wrote two small classes to be able to simplify the code a little bit, influenced by the Adafruit NeoPixel library that I utilise for Arduino-Core MCUs.  I am attaching this at the end for anyone interested. Rudimentary so far, but working :) 

[image]

`class BlockuinoNeoPixelColor {
    public:
        int red;
    int green;
    int blue;
    BlockuinoNeoPixelColor(int in_red, int in_green, int in_blue) {
        red = in_red;
        green = in_green;
        blue = in_blue;
    }
};
class BlockuinoNeoPixel {
    public: int pin;
    int pixels;
    int brightness = 50;
    ManagedBuffer pixelBuffer;
    BlockuinoNeoPixel(int in_pin, int in_pixels) {
        pin = in_pin;
        pixels = in_pixels;
        pixelBuffer = ManagedBuffer(pixels * 3);
        pixelBuffer.fill(0);
    }
    void setPixelColor(int in_pixel, BlockuinoNeoPixelColor color) {
        if (in_pixel <pixels) {
            pixelBuffer[(in_pixel * 3) + 0] = brightness_map(color.green, 0, 255, 0, brightness);
            pixelBuffer[(in_pixel * 3) + 1] = brightness_map(color.red, 0, 255, 0, brightness);
            pixelBuffer[(in_pixel * 3) + 2] = brightness_map(color.blue, 0, 255, 0, brightness);
        }
    }
    void setBrightness(int in_brightness) {
        brightness = in_brightness;
    }
    void show() {
        neopixel_send_buffer(uBit.io.pin[pin], pixelBuffer);
    }
    void clear() {
        pixelBuffer.fill(0);
    }
    void begin() {
        pixelBuffer.fill(0);
    }
    int numPixels() {
        return pixels;
    }
    long brightness_map(long x, long in_min, long in_max, long out_min, long out_max) {
        return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
    }
};
BlockuinoNeoPixel strip(1, 8);`

joachimhs, 2021-08-07:
Is there something similar for the V1 version of the micro:bit?

fabianhugo, 2023-02-13:
Thanks @finneyj for the sample code and @joachimhs for sharing! 
Unfortunately, I am not able to get the neopixel.h functions and the neopixel.h function working. I use a microbit v2 and the microbit-v2-samples repo in the latest version, with build.py.
I successfully built and ran other examples. The only change I made to microbit-v2-samples is in the main.cpp:
```
#include "MicroBit.h"
#include "neopixel.h"
MicroBit uBit;

void neopixel_test()
{
    // Length of neopixel strip
    int length = 8;

    // Colour counter variable (just used to remember what colour we last displayed)
    int c = 0;

    // Create a buffer to hold pixel data of the appropriate size
    ManagedBuffer b(length*3);

    while(1)
    {
        // Clear the buffer to zeroes. You don't need to do this, but it's often convenient.
        b.fill(0);

        // Fill the buffer with some data.
        // Neopixel typically has 3 bytes per pixel, for each primary colour
        // in GRB format (Green, then Red, then Blue).
        for (int i=0; i<length*3; i+=3)
            b[i+c] = 255;

        // Update the neopixel strip on pin P0.
        neopixel_send_buffer(uBit.io.P0, b);

        // Increment our colour counter so that we show a different colour in the next frame.
        c = (c+1) %3;

        // Wait for a second.
        uBit.sleep(1000);
    }
}

int main()
{
    uBit.init();
    while(1){
        neopixel_test();

    }

}
```
I used the GDB debug interface and saw that the micro:bit always goes into a forever loop when trying to execute: `neopixel_send_buffer(uBit.io.P0, b);` Any ideas? 
Can anyone confirm that the neopixel.h file is currently working? 
Thank you, 
Hugo

microbit-carlos, 2023-03-31:
Thanks for the update on your issue @fabianhugo, glad you got it working in the end.
I think we can close this issue as resolved, but if there is anything that still needs looking into please feel free to open the issue 👍
Thanks everyone!

closed 2023-03-31 by microbit-carlos